foxglove/studio #3905: TF visual not updated

tonynajjar, 2022-07-21:
**Description**

- Version: 1.18.0
- OS: Ubuntu 20
- Data source rosbridge:
- ROS distro (if applicable): Galactic

**Steps To Reproduce**

https://drive.google.com/file/d/1HZNp4CoR8tmjaejh8zAa_LqCLI4HqvdI/view?usp=sharing

Please download the video, it was too big to upload here.

On the left: MCAP recording 
On the right: rosbridge connection

In the Mcap recording you can see the TF visual updating
In the rosbridge connection you can see that the TF only updates if I go to another tab and come back

I also show the publishing of the TF in a Raw message panel

jtbandes, 2022-07-21:
Thanks for the video. The problem is visible there, but do you have any tips for how to reproduce it? Is there a node we can download & run (or use something like turtlesim) to reproduce the problem?

tonynajjar, 2022-07-21:
Weird, I just tried on a more e
minimal example and it did not reproduce. I'll look into it tomorrow if I can reproduce it on some open-source node I can share with you

tonynajjar, 2022-07-22:
Finally found out my issue: mismatch between the time of the TF and the time of /clock. TF is in system time and /clock is simulation time. I didn't know that Foxglove uses /clock but it makes sense.

closed 2022-07-22 by jtbandes