foxglove/studio #3907: Self-hosting doesn't work with Chrome

suho0515, 2022-07-21:
**Description**

Hello, I'm trying to run basic container following the manual like below.

`docker run --rm -p "8080:8080" ghcr.io/foxglove/studio:latest`

but when I run the above command, I can only see white screen in chrome browser.

**Chrome**

[image]

But it works well in edge browser.

Microsoft Edge

[image]

Do I need any preliminaries to work this out with chrome?

thanks.

- Version: 1.18
- OS: Window 10
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): None
- ROS distro (if applicable): None

foxymiles, 2022-07-21:
@suho0515 Could you please open the dev console in chrome and refresh the page and see if there are any errors displayed there?

defunctzombie, 2022-07-29:
@suho0515 Were you able to look into this more using dev console in chrome?

suho0515, 2022-07-30:
Sorry for late respose.
I was busy with other projects.

I've check it out, and result is below.

[image]

`<!doctype html><html><head><meta charset="utf-8"><link rel="apple-touch-icon" sizes="180x180" href="[apple-touch-icon.png](http://localhost:8080/apple-touch-icon.png)"/><link rel="icon" type="image/png" sizes="32x32" href="[favicon-32x32.png](http://localhost:8080/favicon-32x32.png)"/><link rel="icon" type="image/png" sizes="16x16" href="[favicon-16x16.png](http://localhost:8080/favicon-16x16.png)"/><title>Foxglove Studio</title><script defer="defer" src="[main.06fdb4eaf19e1fec0d86.js](http://localhost:8080/main.06fdb4eaf19e1fec0d86.js)"></script></head><script>global = globalThis;</script><body><div id="root"></div></body></html>`

Hope to hear any evidence to solve it out.

Thanks again.

defunctzombie, 2022-07-30:
Those are chrome extension errors and nothing specific to do with Studio.

It sounds to me like this is something specific to your environment/setup. Since Edge is working I'd recommend using Edge for now. Since Edge is based on Chrome/v8 you should be able to use all the features of Studio available in chrome.

closed 2022-07-30 by defunctzombie