foxglove/studio #3933: raw messages for diagnostics array message is buggy and some opinions

YairBarzilayBWR, 2022-07-24:
Hi,

I love foxglove its such an important tool and you guys amazing... 

**Bug:**

I upgraded from v1.2.0 to the latest release v1.18.0.

In the newer releases, when using a raw messages panel and the message type is of diagnostic_msgs/DiagnosticArray the 'expand' functionality is broken. i cannot expand the msg as it immediately collapses. 

- Version: 1.18.0
- OS: ubuntu 20
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS1 or bags files.
- ROS distro (if applicable): Noetic

until the 'expand' bug is fixed i unfortunately must downgrade back to v1.2.0 as i deeply rely on diagnostics.

**Some personal opinions:**

Ive noticed a few other things that are a downgrade, at least from my personal user experience:
1. it used to be that when choosing a source and asking for live connection the default would be ROS1, now the default is web socket (personally i dont like this).
2. Currently if you want to access the options of a panel then a new side panel opens on the left side of the entire window, which takes up way more screen real-estate than the previous small dialog box that would open within the panel itself. I like the old version way better as it required less mouse movements. 

Anyways,
Thank you for everything you are doing keep up the great and amazing work!

james-rms, 2022-07-24:
Hi Yair, are you able to share a bag or MCAP file including a DiagnosticArray message that exhibits this problem? I don't see it on the latest release when looking at sample data: 
[image]

foxymiles, 2022-07-25:
@YairBarzilayBWR Can you please also check the settings for your raw messages panel? We added some new auto-expand functionality that might be affecting what you're seeing. You can set this to `off` if you want to control expansion of all fields manually.

[image]

YairBarzilayBWR, 2022-07-27:
Sure Ill check the auto exapand option, did not notice it.

Unfortunately i cannot attach a bag file to git hub and my company does not allow me to supply a link to my onedrive, neither can i upload the file to an external source.

however here is a copy paste of an echo from my terminal, hope this helps.

I'll let you know once i get to upgrading again and checking the auto expand feature.

header: 
  seq: 1092
  stamp: 
    secs: 1655902859
    nsecs: 903993008
  frame_id: ''
status: 
  - 
    level: 0
    name: ''
    message: ''
    hardware_id: ''
    values: 
      - 
        key: "GNSS1_AIDING_STATUS"
        value: "==============================="
      - 
        key: "GNSS#1_integer_fix"
        value: "1"
      - 
        key: "GNSS#1_position_fix"
        value: "1"
      - 
        key: "GNSS#1_diffrential_corrections"
        value: "1"
      - 
        key: "GNSS2_AIDING_STATUS"
        value: "==============================="
      - 
        key: "GNSS#2_integer_fix"
        value: "1"
      - 
        key: "GNSS#2_position_fix"
        value: "1"
      - 
        key: "GNSS#2_diffrential_corrections"
        value: "1"
      - 
        key: "GNSS1_FIX_INFO"
        value: "==============================="
      - 
        key: "GNSS#1_fix_type_ok"
        value: "1"
      - 
        key: "GNSS#1_sbas_used"
        value: "0"
      - 
        key: "GNSS#1_dngss_used"
        value: "0"
      - 
        key: "GNSS2_FIX_INFO"
        value: "==============================="
      - 
        key: "GNSS#2_fix_type_ok"
        value: "1"
      - 
        key: "GNSS#2_sbas_used"
        value: "0"
      - 
        key: "GNSS#2_dngss_used"
        value: "0"
      - 
        key: "NAV_FILTER_STATUS"
        value: "==============================="
      - 
        key: "filter_not_in_error"
        value: "1"
      - 
        key: "filter_stable"
        value: "1"
      - 
        key: "filter_status_ok"
        value: "1"
      - 
        key: "no_heading_warn"
        value: "1"
      - 
        key: "no_imu_bias_warn"
        value: "1"
      - 
        key: "no_lever_arm_warn"
        value: "1"
      - 
        key: "no_position_warn"
        value: "1"
      - 
        key: "no_roll_pitch_warn"
        value: "1"
      - 
        key: "no_time_sync_warn"
        value: "1"
      - 
        key: "no_transform_warn"
        value: "1"
      - 
        key: "RTK_REPORTED_STATUS"
        value: "==============================="
      - 
        key: "dngl_status_ok"
        value: "1"
      - 
        key: "latency_ok"
        value: "1"
      - 
        key: "rtk_status_ok"
        value: "1"
      - 
        key: "DUAL_ANTENNA_STATUS"
        value: "==============================="
      - 
        key: "fix_type_ok"
        value: "1"
      - 
        key: "dt_ok"
        value: "1"
      - 
        key: "uncertainty_under_thresh"
        value: "1"
      - 
        key: "MODULE_STATUS"
        value: "==============================="
      - 
        key: "aiding_status_ok"
        value: "1"
      - 
        key: "dual_antenna_status_ok"
        value: "1"
      - 
        key: "external_ins_status"
        value: "1"
      - 
        key: "external_rtk_status"
        value: "1"
      - 
        key: "fix_info_ok"
        value: "1"
      - 
        key: "internal_ins_status"
        value: "1"
      - 
        key: "internal_rtk_status"
        value: "1"
      - 
        key: "nav_filter_status"
        value: "1"
      - 
        key: "reported_rtk_status"
        value: "1"
      - 
        key: "current_localization_status"
        value: "STATUS_FINE_LOCALIZATION"

amacneil, 2022-07-28:
Thanks for the feedback. We discussed this today and are going to implement some changes described in https://github.com/foxglove/studio/issues/3963.

Regarding your other feedback:

> it used to be that when choosing a source and asking for live connection the default would be ROS1, now the default is web socket (personally i dont like this).

Currently `rosbridge` connection option is at the top of the list because it has the most complete feature support. But you can always use the "recent connections" list to jump to your preferred ROS 1 connection.

[image]

> Currently if you want to access the options of a panel then a new side panel opens on the left side of the entire window, which takes up way more screen real-estate than the previous small dialog [...]

The reason we are moving in this direction is because the previous settings covered up the panel contents. You can see this most clearly in the new `3D Beta` panel. We are still making improvements, so will take your feedback into account - thanks for sharing!