foxglove/studio #3938: Display octomap in FoxeGlove

an99990, 2022-07-25:
Hi,
I would like to visualize octomap grid like rviz2 provides in Foxglove. Currently map panel only displays is looking fro GPS data ?

I provide below an example of map in rviz2.
[image]

thank you

jtbandes, 2022-08-01:
What message types are used to visualize the octomap? Please provide some more information about the data and examples we can look at.

> Currently map panel only displays is looking fro GPS data ?

I think you are using the wrong panel. Try the "3D (Beta)" panel. The "Map" panel is for GPS only.

an99990, 2022-08-02:
hi thank you, i went and found that 3d panel was the correct one. However that panel doesnt seem to be able to display my PointCloud2 coming from the zed2. I did get a tf issue but i changed the frame and the problem went away, but I am still unable to visualize the pointcloud.