foxglove/studio #3942: [ERROR] Exception calling subscribe callback: 'Clock' object has no attribute '_buff'

ljburtz, 2022-07-26:
Hello, 
I get the following error printed at 1000Hz to the terminal of `roslaunch rosbridge_server rosbridge_websocket.launch`
```
2022-07-26 14:16:13+0900 [-] [ERROR] [1658812573.439158, 188.845000]: Exception calling subscribe callback: 'Clock' object has no attribute '_buff'
2022-07-26 14:16:13+0900 [-] [ERROR] [1658812573.440129, 188.846000]: Exception calling subscribe callback: 'Clock' object has no attribute '_buff'
```

- Version: latest (https://studio.foxglove.dev/)
- OS: Ubuntu 22 host (gazebo11 in Ubuntu 20 container)
- Data source rosbridge for ROS1:
- ROS distro (if applicable): ROS noetic

**Steps To Reproduce**
launch gazebo 11, unpause simulation, 
roslaunch rosbridge_server rosbridge_websocket.launch
launch foxglove studio web and select rosbridge connection

**Expected Behavior**
no error for a standard Clock message
https://docs.ros.org/en/noetic/api/rosgraph_msgs/html/msg/Clock.html

**Actual Behavior**
error printed at 1000Hz = clock publish frequency

defunctzombie, 2022-07-29:
Hi - could you make a test case that does not involve Gazebo? Maybe a separate small ros node in python? Reading this issue it isn't clear to me that the problem is on the Studio side vs the ROS side with rosbridge and/or gazebo interaction. I looked through the studio codebase and did not find any references to `_buff`.

defunctzombie, 2022-08-16:
Closing this out since there's been no response. I think this is an issue with rosbridge but would be happy to look if a more specific set of reproduction steps is available.

closed 2022-08-16 by defunctzombie