foxglove/studio #3953: Experimental mcap player: cannot evict the active cache block

james-rms, 2022-07-27:
**Description**

- Version: 1.19.0-dev
- OS: Mac OS 12.4
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): MCAP file generated from `nuscenestomcap` on mini scene dataset
- ROS distro (if applicable): noetic

**Steps To Reproduce**
Not really sure - I tried playing my newly generated MCAP with a 3D beta panel and a raw message panel. Some combination of changing the raw message topic and scrubbing to different parts of the MCAP produced this bug.

https://user-images.githubusercontent.com/18162835/181168272-a24a1e91-d0cf-4d23-9eb3-57ee142642b1.mov

**Expected Behavior**

**Actual Behavior**

amacneil, 2022-07-27:
I got this same error the other day. Couldn't repro it. I was playing around with the 3D (beta?) panel and an image panel. Can't remember if it was an mcap/bag file or sample dataset.

[image]

amacneil, 2022-07-27:
This error should be reported to sentry, it is not user-fixable.

defunctzombie, 2022-07-27:
Looking into this now.

@amacneil I believe this error is also reported to sentry.

PR #3975 (2022-07-28): Reset total size when clearing CachingIterableSource cache

**User-Facing Changes**
Experimental player users should no longer see errors for "cannot evict active cache block". 

**Description**
When the CachingIterableSource cache is cleared, the total size was not being cleared. This resulted in a cache that thought it was full when it wasn't. This triggered code paths to purge blocks including the active block that we just created.

This change cleares the total size when clearing the cache, re-organizes when blocks are ended due to them being over size, and adds tests for both behaviors.

 [... cut]
commit 28467cbd (2022-07-28): Reset total size when clearing CachingIterableSource cache (#3975)

When the CachingIterableSource cache is cleared, the total size was not being cleared. This resulted in a cache that thought it was full when it wasn't. This triggered code paths to purge blocks including the active block that we just created.

This change cleares the total size when clearing the cache, re-organizes when blocks are ended due to them being over size, and adds tests for both behaviors.

Fixes: #3953
closed 2022-07-28 by defunctzombie