foxglove/studio #3954: 3D (Beta) Panel error when MarkerArray points array becomes empty

GreatAlexander, 2022-07-27:
**Description**
The 3D (Beta) Panel stops working and becomes un-responsive if a visualised Marker Array topic transitions from having a non-empty points array -> empty points array (Issue does not occur if marker type does not use the points array e.g. CUBE instead of LINE_STRIP as demonstrated below)

- Version:
- OS: Ubuntu 20.04
- Data source: ROS 1 Rosbag
- ROS distro: ROS Noetic

**Steps To Reproduce**
1. In a Node Playground, produce a marker array of a marker type that requires an array of points (i.e. MarkerTypes.LINE_STRIP)
2. Setup a 3D (Beta) panel that displays that marker array topic
3. In the Node Playground, set it up so the marker array list of points displayed is non-zero (>1 for viz to work)
4. Modify the Node Playground code so mid-ROSbag the list of points becomes empty [I tried to reproduce this easily by a Global Variable but that did not reproduce this issue. Best is to make the point array change live given a trigger from data in the ROSbag].
5. Play the rosbag, make sure visualisation topic is being displayed.

Here are some example pictures showing a line strip marker array (The one reproducing this issue) and another marker for reference (To note when the 3D Beta panel stops working). On the right is the input marker array msg reproducing this issue.
- Before points array becomes empty
[image]
- Points array becomes empty (Note 3D Beta panel stops working)
[image]
- If a new 3D (Beta) panel is created now, this error is shown
[image]
- Points array is non-empty again (Note 3D Beta panel has not recovered, whereas the 3D panel is still working as expected)
[image]

**Expected Behavior**
3D (Beta) panel should behave like the 3D panel and not stop working. It should flag that the marker array contains no points and thus cannot be displayed.

**Actual Behavior**
3D (Beta) panel stops working and has to be Reset. If reset while the empty points marker array msg is still selected it will display

jhurliman, 2022-08-05:
I was not able to reproduce this bug with a CUBE_LIST. I will continue to investigate, meanwhile can you tell me if you experience this issue using the attached file?

[filled-empty-filled.mcap.zip](https://github.com/foxglove/studio/files/9264878/filled-empty-filled.mcap.zip)

jhurliman, 2022-08-05:
Here is one using LINE_STRIP instead of cube (ignore the topic name, the marker type is set to 5 / LINE_STRIP).

[filled-empty-filled2.mcap.zip](https://github.com/foxglove/studio/files/9264910/filled-empty-filled2.mcap.zip)

GreatAlexander, 2022-08-08:
Thanks for these mcaps to try to reproduce @jhurliman .
Indeed I can confirm neither of these mcap files reproduce the issue for me.
Could you do the same test with LINE_STRIP instead?
i.e. go from a LINE_STRIP of 4096 points to 0 to 4096 again?

defunctzombie, 2022-08-09:
@GreatAlexander Are you able to provide a bag file or mcap file that we can play showing the issue reproduced?

It might be that this is specific to user script (formerly node playground) interaction.

jtbandes, 2022-08-10:
Thanks for the report, and for sharing the data/layout to reproduce! I have a PR to fix the issue at: #4081

Confirmed that John's second sample data was using type 5 (line list) instead of 4 (line strip), and only line strip markers suffer from this issue.

PR #4081 (2022-08-10): [New3D] fix crash when updating line strip to be empty
**User-Facing Changes**
Fixed a crash in the 3D (Beta) when a line strip marker was updated to have 0 points.

**Description**
Fixes #3954.

THREE.LineGeometry.setPositions does not handle an empty array: https://github.com/mrdoob/three.js/blob/38273fb9b2acde3f6046301757d9f1e5428f7d9d/examples/jsm/lines/LineGeometry.js#L15-L20

Avoid the crash by hiding the renderables and adding an early return when the new length is zero.
commit 506969cb (2022-08-10): [New3D] fix crash when updating line strip to be empty (#4081)

**User-Facing Changes**
Fixed a crash in the 3D (Beta) when a line strip marker was updated to have 0 points.

**Description**
Fixes #3954.

THREE.LineGeometry.setPositions does not handle an empty array: https://github.com/mrdoob/three.js/blob/38273fb9b2acde3f6046301757d9f1e5428f7d9d/examples/jsm/lines/LineGeometry.js#L15-L20

Avoid the crash by hiding the renderables and adding an early return when the new length is zero.
closed 2022-08-10 by jtbandes