foxglove/studio #3956: Toggle panel settings on Settings gear click

rgov, 2022-07-27:
The Settings button (gear icon) in the panel toolbar shows the panel settings sidebar, but cannot be used to hide it. If I mis-click it, it's a long trek for the mouse to hide the settings pane.

Desired logic:

1. If the Settings pane is not shown, open it and show the target panel's settings.
2. If the Settings pane is open and showing another panel's settings, show the target panel's settings.
3. If the Settings pane is open and already showing the target panel's settings, hide the pane.

Foxglove Studio 1.16.0

defunctzombie, 2022-07-27:
👍 

Question on:
> 2\. If the Settings pane is open and showing another panel's settings, show the target panel's settings.

The panel settings always shows the target panel's settings - is there a situation where it is not?

rgov, 2022-07-27:
> The panel settings always shows the target panel's settings - is there a situation where it is not?

It follows focus correctly. I just want to clarify that the first click on a non-focused panel's Settings button should never hide the sidebar.

amacneil, 2022-07-27:
I think it would be interesting to try the functionality as described by @rgov.

My concern is that sometimes I click the Settings gear icon just to make sure I've got the right panel settings up (e.g. if I have two image panels, and click the gear icon to make sure I'm looking at the right settings before I change it).

FYI we did recently add support for Cmd+B / Ctrl+B keyboard shortcut to hide the sidebar (https://github.com/foxglove/studio/pull/3939).

jtbandes, 2022-08-01:
As I commented on the equivalent customer request in Slack:

> this mostly seems useful for panels on the rightmost edge of your layout. for panels elsewhere, the button will move when you click it the first time, so clicking it a second time isn’t much easier than clicking on the sidebar icon to close the sidebar

rgov, 2022-08-01:
Agreed! But at least then I can keep my eyes on the panel I'm interested in rather than switching to chase down the sidebar. But I'll have to learn to react with <kbd>⌘-B</kbd> instead.

defunctzombie, 2022-08-16:
We gave this a try in https://github.com/foxglove/studio/pull/4016 but found the behavior was not to our liking. Thanks for the suggestion but for now we will stick with the current behavior.

closed 2022-08-16 by defunctzombie