foxglove/studio #3967: Data source topic list should show all topics, including  unsupported ones

amacneil, 2022-07-28:
**Description**
I downloaded an open source db3 file from [here](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/).

When I open it in Foxglove Studio, the "Topics" list is missing a few topics:

[image]

They do show up in the "Problems" tab:

[image]

I think we should show all topics in the Topics list, even if we are missing a message definition. We could still list the topic name, schema name, and publish frequency. We could also show an error inline next to the topic (similar to sidebar settings errors).

amacneil, 2022-07-28:
cc @defunctzombie for feedback

defunctzombie, 2022-07-30:
I think it's up to the data source to emit the topics it wants to make available. In this case the db3 source I probably not adding these topics to its list since it doesn't have definitions for them.

Its a simple change to switch up the logic here: https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/IterablePlayer/rosdb3/RosDb3IterableSource.worker.ts#L74-L80

I don't have super strong feelings either way but seems fine to show the user the topic and datatype and change the warnings to something more actionable.

PR #4003 (2022-07-31): Update RosDb3IterableSource to list all topics

**User-Facing Changes**
Data source info shows all topics found in .db3 files - even those we cannot deserialize.

**Description**
Even if a topic cannot be deserialized (due to missing message definitions), we still show it in the topics results so the user knows it is present in their file.

Fixes: #3967
commit 3ffcfb58 (2022-07-31): Update RosDb3IterableSource to list all topics (#4003)

**User-Facing Changes**
Data source info shows all topics found in .db3 files - even those we cannot deserialize.

**Description**
Even if a topic cannot be deserialized (due to missing message definitions), we still show it in the topics results so the user knows it is present in their file.

Fixes: #3967
closed 2022-07-31 by defunctzombie