foxglove/studio #3991: Plot: Option to set "x min" and "x max"

amacneil, 2022-07-29:
Ability to set "x min" and "x max" in addition to "y min"m and "y max" (already supported).

This is important for consistent/comparable plots.

It could be added to the plot settings:

[image]

(extracted from https://github.com/foxglove/studio/issues/3990)

foxymiles, 2022-08-02:
@amacneil Would these new settings also go under `Time series only`?

amacneil, 2022-08-02:
I think we could create some new groups for `X axis` and `Y axis` (or google sheets calls them `Horizontal axis` and `Vertical axis`):

[image]

defunctzombie, 2022-08-16:
Might be nice to also hide settings if they are not relevant. It was harder to do this before but seems sensible now.

PR #4145 (2022-08-25): Add manual limits for x axis
**User-Facing Changes**
This adds new plot panel settings to manually specify the min and max X axis values.

**Description**
This adds new settings and new logic to clamp the X axis when both min and max are specified.

It also separates X & Y related settings into their own nodes in the settings sidebar.

[image]

Fixes #3991
Fixes #4251
closed 2022-08-25 by foxymiles