foxglove/studio #4004: Unable to play rosbag file from specific seek time

saumilsdk, 2022-07-30:
ROSBAG file is uploaded to Azure ADLS Gen2 storage account and URL is constructed with the seek time as per the foxglove documentation by using "time" field. ROSBAG file always plays from start and doesn't start from the time provided.

E.g.
https://xyz.blob.core.windows.net/sensor.bag?time=2022-07-20T11:16:50+05:30&se=2022-07-29T11%3A00%3A40Z&sp=r&sv=2021-06-08&sr=b&sig=XXXX

Tried with encoded values also but doesn't work:

https://xyz.blob.core.windows.net/sensor.bag?time=2022-07-20T11%3A16%3A50%2B05%3A30

foxymiles, 2022-08-01:
Can you please share the complete URL you're using to open studio? You need to make sure that the `time` parameter isn't being encoded as part your data source url.

saumilsdk, 2022-08-02:
@foxymiles Complete URL contains storage account name and signature so will strip that part. 

https://<storageAccount>.blob.core.windows.net/rawdata/sensor.bag?time=2022-07-20T11:16:50+05:30&se=2022-07-29T11%3A00%3A40Z&sp=r&sv=2021-06-08&sr=b&sig=nTS5dnmSRmXYZKWkJPsJQsPkIx%2BxbJyYk2c%3D

foxymiles, 2022-08-02:
Sorry I should have been more clear. Can you show me the exact url (without name and signature) you're using to open studio? It should be similar to the URLs described here:

https://foxglove.dev/docs/studio/building-and-sharing-links

Like:

https://studio.foxglove.dev/?ds=ros1-remote-bagfile&ds.url=https://storage.googleapis.com/foxglove-public-assets/demo.bag 

or

foxglove://open?ds=ros1-socket&ds.url=localhost:11311

defunctzombie, 2022-08-17:
Closing this since there's been no follow-up response. I've not been able to reproduce this using our sample data which is also serviced from object storage. Here's a url that includes a specific timestamp and when opening it, Studio seeks to that time:

https://studio.foxglove.dev/?ds=sample-nuscenes&layoutId=aa1f5c51-c256-44aa-827b-ca49ad734dc4&time=2018-07-24T03%3A28%3A52.477934361Z

If you are able to share a specific link to reproduce the problem then we can take a look.

closed 2022-08-17 by defunctzombie