foxglove/studio #4005: Support rosbridge Fragmentation

nathanmargaglio, 2022-07-30:
The [rosbridge protocol provides the ability to fragment messages](https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/ROSBRIDGE_PROTOCOL.md#311-fragmentation--fragment--experimental).  This feature, however, doesn't appear implemented on studio's side.  When connecting to Foxglove with rosbridge with fragmentation enabled, the fragmented messages appear to come through correctly (as per the Network tab in Developer Tools), but Foxglove doesn't seem to be interpreting them.

Is there any plans to implement this?  If not, is there some direction that can be provided in terms of getting this implemented?  We are forced to use fragmented messages due to our infrastructure constraints, which prevents us from using Foxglove (which would be ideal).  We'd be happy to work on this implementation if it's feasible.  Thanks!

jtbandes, 2022-08-01:
Our rosbridge connection feature uses [roslibjs](https://github.com/RobotWebTools/roslibjs) under the hood, so that would be the place where this needs to be implemented. I would recommend filing an issue in that repo (if there isn't one already) and making a PR there!

defunctzombie, 2022-11-04:
Thanks for the feature request however we have no plans to support the fragmentation feature of rosbridge. We are actively working on a c++ replacement for rosbridge using the foxglove websocket protocol. This implementation resolves some long standing performance bottlenecks and limitations we've experienced with the python version of rosbridge. Once that is available we'd be happy to have you give that a test and see how that works in your environment or whether there are missing aspects of the foxglove websocket protocol.

closed 2022-11-04 by defunctzombie