foxglove/studio #4006: emits git-lfs "Smudge error" when merging from main branch

lizhucheng, 2022-08-01:
**Description**
my repository was forked from this and git HEAD at commit ed174aa0 (near version 1.6.0.)
after  git merge  by following shell scripts, I got a git-lfs error. it sences some files didn't commit correctly. 

sh codes: 
```
git remote add main https://github.com/foxglove/studio
git fetch main
git merge main/main
```

emit error:
```
更新 ed174aa0..3ffcfb58
Downloading packages/studio-base/src/panels/Indicator/thumbnail.png (3.5 KB)
Error downloading object: packages/studio-base/src/panels/Indicator/thumbnail.png (7ad2149): Smudge error: Error downloading packages/studio-base/src/panels/Indicator/thumbnail.png (7ad2149c9cac8f92584e252d8ee83b991fbcc607197c8ee1a25265059adf37ea): [7ad2149c9cac8f92584e252d8ee83b991fbcc607197c8ee1a25265059adf37ea] Object does not exist on the server or you don't have permissions to access it: [404] Object does not exist on the server or you don't have permissions to access it
```

- Version: current foxglove version tag 1.6.0
- OS: linux ubuntu 20.04 focal / macos

**Steps To Reproduce**

git remote add main https://github.com/foxglove/studio
git fetch main
git merge main/main

**Expected Behavior**
git merge successful

**Actual Behavior**
git merge failed

lizhucheng, 2022-08-01:
when I exec the same codes in a mac computer without git-lfs extention, everything is ok!

foxymiles, 2022-08-01:
I just tried doing a fresh checkout of the repo and the `packages/studio-base/src/panels/Indicator/thumbnail.png` file seems to be present as expected so I don't the issue is on the repo side. Maybe you've gotten git into some uncertain state?

What output do you get when running `git lfs ls-files --all`?

defunctzombie, 2022-08-01:
This sounds like something specific to your environment and git as we are able to checkout the code without issues. Closing since it isn't a Studio bug. Maybe try deleting your repo and getting a fresh clone.

closed 2022-08-01 by defunctzombie

lizhucheng, 2022-08-02:
thanks.
git head at commit ed174aa02d
the logs of `git lfs ls-files --all`   like these:
> 

997be80aa3 * .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0d07c9d5f0 * .yarn/plugins/@yarnpkg/plugin-typescript.cjs
945c688bc7 * .yarn/plugins/@yarnpkg/plugin-version.cjs
3bd7eae0e1 * .yarn/plugins/@yarnpkg/plugin-workspace-tools.cjs
5e197ae112 * .yarn/releases/yarn-3.1.0.cjs
3209a6de07 * packages/studio-base/src/panels/GlobalVariableSlider/thumbnail.png
6ac2af4b6c * packages/studio-base/src/panels/ImageView/thumbnail.png
a9bdd84cc8 * packages/studio-base/src/panels/Log/thumbnail.png
9e9ffa47bb * packages/studio-base/src/panels/Map/thumbnail.png
82cc12c7fe * packages/studio-base/src/panels/NodePlayground/thumbnail.png
2bd7e8e121 * packages/studio-base/src/panels/Plot/thumbnail.png
4a06bcddff * packages/studio-base/src/panels/Publish/thumbnail.png
6a75dd3633 * packages/studio-base/src/panels/RawMessages/thumbnail.png
3599ac4b9b * packages/studio-base/src/panels/SourceInfo/thumbnail.png
bb750517f6 * packages/studio-base/src/panels/StateTransitions/thumbnail.png
d52665ee8c * packages/studio-base/src/panels/Tab/thumbnail.png
636064cd92 * packages/studio-base/src/panels/Table/thumbnail.png
d395412241 * packages/studio-base/src/panels/Teleop/thumbnail.png
52705fc66f * packages/studio-base/src/panels/ThreeDimensionalViz/commands/404.glb
30d6f20606 * packages/studio-base/src/panels/ThreeDimensionalViz/thumbnail.png
7652dd890d * packages/studio-base/src/panels/ThreeDimensionalViz/utils/textureAtlasData.bin
76ac540ab6 * packages/studio-base/src/panels/TopicGraph/thumbnail.png
8ef9bce9f9 * packages/studio-base/src/panels/URDFViewer/thumbnail.png
56c9547056 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-status.png
bf3735b890 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-summary.png
900058dffa * packages/studio-base/src/styles/assets/Inter-Italic.woff2
1b498b959e * packages/studio-base/src/styles/assets/Inter-Medium.woff2
816008586a * packages/studio-base/src/styles/assets/Inter-MediumItalic.woff2
d612f1212b * packages/studio-base/src/styles/assets/Inter-Regular.woff2
152261291c * packages/studio-base/src/styles/assets/Inter-SemiBold.woff2
3b6df7d05e * packages/studio-base/src/styles/assets/Inter-SemiBoldItalic.woff2
b705f93a7d * packages/studio-base/src/styles/assets/PlexMono-Bold.woff2
d2b4296cf2 * packages/studio-base/src/styles/assets/PlexMono-BoldItalic.woff2
5dac9624f7 * packages/studio-base/src/styles/assets/PlexMono-Italic.woff2
a45517d83e * packages/studio-base/src/styles/assets/PlexMono.woff2
5274d7c777 * packages/studio-base/src/test/fixtures/demo-shuffled.bag
c57db94780 * packages/studio-base/src/test/fixtures/example-bz2.bag
6f8b495a72 * packages/studio-base/src/test/fixtures/example.bag
c14768c26a * resources/dmg-background/background.afdesign
88f5bc1922 * resources/dmg-background/background.png
670c2a35de * resources/dmg-background/background@2x.png
53c86f113b * resources/icon/BagIcon.afdesign
f9118bab41 * resources/icon/BagIcon.ico
3f2804d9e2 * resources/icon/BagIcon.png
b7679ce2d3 * resources/icon/FoxeIcon.ico
5d0e027393 * resources/icon/FoxeIcon.png
0aacdb5120 * resources/icon/McapIcon.ico
66597c2cfe * resources/icon/McapIcon.png
7638e7b295 * resources/icon/URDFIcon.ico
224461947d * resources/icon/URDFIcon.png
e09b2bcd7d * resources/icon/XacroIcon.ico
bd7a65b400 * resources/icon/XacroIcon.png
d47b301959 * resources/icon/icon.afdesign
40cab45a99 * resources/icon/icon.icns
b5bbf7b457 * resources/icon/icon.png
5ee889c6b9 * resources/screenshot.png
fbbb03abb3 * web/public/apple-touch-icon.png
2a78a241a8 * web/public/favicon-16x16.png
983742d489 * web/public/favicon-32x32.png
ae55fb5a76 * web/public/favicon.ico
**7ad2149c9c - packages/studio-base/src/panels/Indicator/thumbnail.png**
ac8e6ac602 - packages/studio-base/src/panels/Gauge/thumbnail.png
705a1b8391 - packages/studio-base/src/panels/ThreeDimensionalViz/commands/CarModel/carModel.glb
dd075b46a9 - .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
d6c3c6912c - .yarn/plugins/@yarnpkg/plugin-typescript.cjs
[... 12 lines cut ...]
9dd75dfd10 - .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0779a5d254 - .yarn/plugins/@yarnpkg/plugin-typescript.cjs
319f6c5f50 - .yarn/plugins/@yarnpkg/plugin-version.cjs
8feb0398db - .yarn/releases/yarn-2.4.1.cjs
3f953a17a4 - resources/icon/BagIcon.afdesign
3cfc3a079c - resources/icon/BagIcon.ico
0c93473499 - resources/icon/BagIcon.png
5ddcf4633a - resources/icon/URDFIcon.ico
f2f3c01b81 - resources/icon/URDFIcon.png
68f943dbcd - resources/icon/XacroIcon.ico
7e10eefc36 - resources/icon/XacroIcon.png
9b5c31c1ed - resources/icon/icon.afdesign
8deb6bd2bf - resources/icon/icon.icns
bf5dd2f409 - resources/icon/icon.png
5ba954dfd5 - packages/rosmsg-bobject/wasm/translator.wasm
a9b84d4ba3 - packages/rosmsg-bobject/wasm/translator.wasm.map
5c3f78eb36 - app/components/WssErrorModal.png
87011166ab - app/styles/assets/latin-roboto-mono-bold-italic.woff2
a05c6840f7 - app/styles/assets/latin-roboto-mono-bold.woff2
559c316bda - app/styles/assets/latin-roboto-mono-italic.woff2
3a64ada5df - app/styles/assets/latin-roboto-mono.woff2
4e5a0f9e9a - resources/screenshot.jpg
b19342a0e7 - packages/rosmsg-bobject/bin/translator.wasm
441e61f2bf - packages/rosmsg-bobject/bin/translator.wasm.map
20e1fc9067 - .yarn/releases/yarn-berry.cjs

lizhucheng, 2022-08-02:
> I just tried doing a fresh checkout of the repo and the `packages/studio-base/src/panels/Indicator/thumbnail.png` file seems to be present as expected so I don't think the issue is on the repo side. Maybe you've gotten git into some uncertain state?
> 
> What output do you get when running `git lfs ls-files --all`?

thanks.
git head at commit https://github.com/foxglove/studio/commit/ed174aa02dd0ff8fd1ec056b7ae0e059fdf5896c
the logs of git lfs ls-files --all like these: (my git-lfs version is 3.1.4, maybe it the reason, a broken version of git-lfs? can you tell me what's the version of git-lfs you use.)

997be80aa3 * .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0d07c9d5f0 * .yarn/plugins/@yarnpkg/plugin-typescript.cjs
945c688bc7 * .yarn/plugins/@yarnpkg/plugin-version.cjs
3bd7eae0e1 * .yarn/plugins/@yarnpkg/plugin-workspace-tools.cjs
5e197ae112 * .yarn/releases/yarn-3.1.0.cjs
3209a6de07 * packages/studio-base/src/panels/GlobalVariableSlider/thumbnail.png
6ac2af4b6c * packages/studio-base/src/panels/ImageView/thumbnail.png
a9bdd84cc8 * packages/studio-base/src/panels/Log/thumbnail.png
9e9ffa47bb * packages/studio-base/src/panels/Map/thumbnail.png
82cc12c7fe * packages/studio-base/src/panels/NodePlayground/thumbnail.png
2bd7e8e121 * packages/studio-base/src/panels/Plot/thumbnail.png
4a06bcddff * packages/studio-base/src/panels/Publish/thumbnail.png
6a75dd3633 * packages/studio-base/src/panels/RawMessages/thumbnail.png
3599ac4b9b * packages/studio-base/src/panels/SourceInfo/thumbnail.png
bb750517f6 * packages/studio-base/src/panels/StateTransitions/thumbnail.png
d52665ee8c * packages/studio-base/src/panels/Tab/thumbnail.png
636064cd92 * packages/studio-base/src/panels/Table/thumbnail.png
d395412241 * packages/studio-base/src/panels/Teleop/thumbnail.png
52705fc66f * packages/studio-base/src/panels/ThreeDimensionalViz/commands/404.glb
30d6f20606 * packages/studio-base/src/panels/ThreeDimensionalViz/thumbnail.png
7652dd890d * packages/studio-base/src/panels/ThreeDimensionalViz/utils/textureAtlasData.bin
76ac540ab6 * packages/studio-base/src/panels/TopicGraph/thumbnail.png
8ef9bce9f9 * packages/studio-base/src/panels/URDFViewer/thumbnail.png
56c9547056 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-status.png
bf3735b890 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-summary.png
900058dffa * packages/studio-base/src/styles/assets/Inter-Italic.woff2
1b498b959e * packages/studio-base/src/styles/assets/Inter-Medium.woff2
816008586a * packages/studio-base/src/styles/assets/Inter-MediumItalic.woff2
d612f1212b * packages/studio-base/src/styles/assets/Inter-Regular.woff2
152261291c * packages/studio-base/src/styles/assets/Inter-SemiBold.woff2
3b6df7d05e * packages/studio-base/src/styles/assets/Inter-SemiBoldItalic.woff2
b705f93a7d * packages/studio-base/src/styles/assets/PlexMono-Bold.woff2
d2b4296cf2 * packages/studio-base/src/styles/assets/PlexMono-BoldItalic.woff2
5dac9624f7 * packages/studio-base/src/styles/assets/PlexMono-Italic.woff2
a45517d83e * packages/studio-base/src/styles/assets/PlexMono.woff2
5274d7c777 * packages/studio-base/src/test/fixtures/demo-shuffled.bag
c57db94780 * packages/studio-base/src/test/fixtures/example-bz2.bag
6f8b495a72 * packages/studio-base/src/test/fixtures/example.bag
c14768c26a * resources/dmg-background/background.afdesign
88f5bc1922 * resources/dmg-background/background.png
670c2a35de * resources/dmg-background/background@2x.png
53c86f113b * resources/icon/BagIcon.afdesign
f9118bab41 * resources/icon/BagIcon.ico
3f2804d9e2 * resources/icon/BagIcon.png
b7679ce2d3 * resources/icon/FoxeIcon.ico
5d0e027393 * resources/icon/FoxeIcon.png
0aacdb5120 * resources/icon/McapIcon.ico
66597c2cfe * resources/icon/McapIcon.png
7638e7b295 * resources/icon/URDFIcon.ico
224461947d * resources/icon/URDFIcon.png
e09b2bcd7d * resources/icon/XacroIcon.ico
bd7a65b400 * resources/icon/XacroIcon.png
d47b301959 * resources/icon/icon.afdesign
40cab45a99 * resources/icon/icon.icns
b5bbf7b457 * resources/icon/icon.png
5ee889c6b9 * resources/screenshot.png
fbbb03abb3 * web/public/apple-touch-icon.png
2a78a241a8 * web/public/favicon-16x16.png
983742d489 * web/public/favicon-32x32.png
ae55fb5a76 * web/public/favicon.ico
7ad2149c9c - packages/studio-base/src/panels/Indicator/thumbnail.png
ac8e6ac602 - packages/studio-base/src/panels/Gauge/thumbnail.png
[... 15 lines cut ...]
9dd75dfd10 - .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0779a5d254 - .yarn/plugins/@yarnpkg/plugin-typescript.cjs
319f6c5f50 - .yarn/plugins/@yarnpkg/plugin-version.cjs
8feb0398db - .yarn/releases/yarn-2.4.1.cjs
3f953a17a4 - resources/icon/BagIcon.afdesign
3cfc3a079c - resources/icon/BagIcon.ico
0c93473499 - resources/icon/BagIcon.png
5ddcf4633a - resources/icon/URDFIcon.ico
f2f3c01b81 - resources/icon/URDFIcon.png
68f943dbcd - resources/icon/XacroIcon.ico
7e10eefc36 - resources/icon/XacroIcon.png
9b5c31c1ed - resources/icon/icon.afdesign
8deb6bd2bf - resources/icon/icon.icns
bf5dd2f409 - resources/icon/icon.png
5ba954dfd5 - packages/rosmsg-bobject/wasm/translator.wasm
a9b84d4ba3 - packages/rosmsg-bobject/wasm/translator.wasm.map
5c3f78eb36 - app/components/WssErrorModal.png
87011166ab - app/styles/assets/latin-roboto-mono-bold-italic.woff2
a05c6840f7 - app/styles/assets/latin-roboto-mono-bold.woff2
559c316bda - app/styles/assets/latin-roboto-mono-italic.woff2
3a64ada5df - app/styles/assets/latin-roboto-mono.woff2
4e5a0f9e9a - resources/screenshot.jpg
b19342a0e7 - packages/rosmsg-bobject/bin/translator.wasm
441e61f2bf - packages/rosmsg-bobject/bin/translator.wasm.map
20e1fc9067 - .yarn/releases/yarn-berry.cjs

lizhucheng, 2022-08-02:
> I just tried doing a fresh checkout of the repo and the `packages/studio-base/src/panels/Indicator/thumbnail.png` file seems to be present as expected so I don't think the issue is on the repo side. [...]

thanks.
git head at commit https://github.com/foxglove/studio/commit/ed174aa02dd0ff8fd1ec056b7ae0e059fdf5896c
the logs of git lfs ls-files --all like these: (my git-lfs version is 3.1.4, maybe it the reason, a broken version of git-lfs? can you tell me what's the version of git-lfs you use.)

997be80aa3 * .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0d07c9d5f0 * .yarn/plugins/@yarnpkg/plugin-typescript.cjs
945c688bc7 * .yarn/plugins/@yarnpkg/plugin-version.cjs
3bd7eae0e1 * .yarn/plugins/@yarnpkg/plugin-workspace-tools.cjs
5e197ae112 * .yarn/releases/yarn-3.1.0.cjs
3209a6de07 * packages/studio-base/src/panels/GlobalVariableSlider/thumbnail.png
6ac2af4b6c * packages/studio-base/src/panels/ImageView/thumbnail.png
a9bdd84cc8 * packages/studio-base/src/panels/Log/thumbnail.png
9e9ffa47bb * packages/studio-base/src/panels/Map/thumbnail.png
82cc12c7fe * packages/studio-base/src/panels/NodePlayground/thumbnail.png
2bd7e8e121 * packages/studio-base/src/panels/Plot/thumbnail.png
4a06bcddff * packages/studio-base/src/panels/Publish/thumbnail.png
6a75dd3633 * packages/studio-base/src/panels/RawMessages/thumbnail.png
3599ac4b9b * packages/studio-base/src/panels/SourceInfo/thumbnail.png
bb750517f6 * packages/studio-base/src/panels/StateTransitions/thumbnail.png
d52665ee8c * packages/studio-base/src/panels/Tab/thumbnail.png
636064cd92 * packages/studio-base/src/panels/Table/thumbnail.png
d395412241 * packages/studio-base/src/panels/Teleop/thumbnail.png
52705fc66f * packages/studio-base/src/panels/ThreeDimensionalViz/commands/404.glb
30d6f20606 * packages/studio-base/src/panels/ThreeDimensionalViz/thumbnail.png
7652dd890d * packages/studio-base/src/panels/ThreeDimensionalViz/utils/textureAtlasData.bin
76ac540ab6 * packages/studio-base/src/panels/TopicGraph/thumbnail.png
8ef9bce9f9 * packages/studio-base/src/panels/URDFViewer/thumbnail.png
56c9547056 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-status.png
bf3735b890 * packages/studio-base/src/panels/diagnostics/thumbnails/diagnostic-summary.png
900058dffa * packages/studio-base/src/styles/assets/Inter-Italic.woff2
1b498b959e * packages/studio-base/src/styles/assets/Inter-Medium.woff2
816008586a * packages/studio-base/src/styles/assets/Inter-MediumItalic.woff2
d612f1212b * packages/studio-base/src/styles/assets/Inter-Regular.woff2
152261291c * packages/studio-base/src/styles/assets/Inter-SemiBold.woff2
3b6df7d05e * packages/studio-base/src/styles/assets/Inter-SemiBoldItalic.woff2
b705f93a7d * packages/studio-base/src/styles/assets/PlexMono-Bold.woff2
d2b4296cf2 * packages/studio-base/src/styles/assets/PlexMono-BoldItalic.woff2
5dac9624f7 * packages/studio-base/src/styles/assets/PlexMono-Italic.woff2
a45517d83e * packages/studio-base/src/styles/assets/PlexMono.woff2
5274d7c777 * packages/studio-base/src/test/fixtures/demo-shuffled.bag
c57db94780 * packages/studio-base/src/test/fixtures/example-bz2.bag
6f8b495a72 * packages/studio-base/src/test/fixtures/example.bag
c14768c26a * resources/dmg-background/background.afdesign
88f5bc1922 * resources/dmg-background/background.png
670c2a35de * resources/dmg-background/background@2x.png
53c86f113b * resources/icon/BagIcon.afdesign
f9118bab41 * resources/icon/BagIcon.ico
3f2804d9e2 * resources/icon/BagIcon.png
b7679ce2d3 * resources/icon/FoxeIcon.ico
5d0e027393 * resources/icon/FoxeIcon.png
0aacdb5120 * resources/icon/McapIcon.ico
66597c2cfe * resources/icon/McapIcon.png
7638e7b295 * resources/icon/URDFIcon.ico
224461947d * resources/icon/URDFIcon.png
e09b2bcd7d * resources/icon/XacroIcon.ico
bd7a65b400 * resources/icon/XacroIcon.png
d47b301959 * resources/icon/icon.afdesign
40cab45a99 * resources/icon/icon.icns
b5bbf7b457 * resources/icon/icon.png
5ee889c6b9 * resources/screenshot.png
fbbb03abb3 * web/public/apple-touch-icon.png
2a78a241a8 * web/public/favicon-16x16.png
983742d489 * web/public/favicon-32x32.png
ae55fb5a76 * web/public/favicon.ico
**7ad2149c9c - packages/studio-base/src/panels/Indicator/thumbnail.png**
ac8e6ac602 - packages/studio-base/src/panels/Gauge/thumbnail.png
705a1b8391 - packages/studio-base/src/panels/ThreeDimensionalViz/commands/CarModel/carModel.glb
dd075b46a9 - .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
[... 13 lines cut ...]
9dd75dfd10 - .yarn/plugins/@yarnpkg/plugin-interactive-tools.cjs
0779a5d254 - .yarn/plugins/@yarnpkg/plugin-typescript.cjs
319f6c5f50 - .yarn/plugins/@yarnpkg/plugin-version.cjs
8feb0398db - .yarn/releases/yarn-2.4.1.cjs
3f953a17a4 - resources/icon/BagIcon.afdesign
3cfc3a079c - resources/icon/BagIcon.ico
0c93473499 - resources/icon/BagIcon.png
5ddcf4633a - resources/icon/URDFIcon.ico
f2f3c01b81 - resources/icon/URDFIcon.png
68f943dbcd - resources/icon/XacroIcon.ico
7e10eefc36 - resources/icon/XacroIcon.png
9b5c31c1ed - resources/icon/icon.afdesign
8deb6bd2bf - resources/icon/icon.icns
bf5dd2f409 - resources/icon/icon.png
5ba954dfd5 - packages/rosmsg-bobject/wasm/translator.wasm
a9b84d4ba3 - packages/rosmsg-bobject/wasm/translator.wasm.map
5c3f78eb36 - app/components/WssErrorModal.png
87011166ab - app/styles/assets/latin-roboto-mono-bold-italic.woff2
a05c6840f7 - app/styles/assets/latin-roboto-mono-bold.woff2
559c316bda - app/styles/assets/latin-roboto-mono-italic.woff2
3a64ada5df - app/styles/assets/latin-roboto-mono.woff2
4e5a0f9e9a - resources/screenshot.jpg
b19342a0e7 - packages/rosmsg-bobject/bin/translator.wasm
441e61f2bf - packages/rosmsg-bobject/bin/translator.wasm.map
20e1fc9067 - .yarn/releases/yarn-berry.cjs

lizhucheng, 2022-08-02:
> This sounds like something specific to your environment and git as we are able to checkout the code without issues. Closing since it isn't a Studio bug. Maybe try deleting your repo and getting a fresh clone.

thanks for your reply. 
a fresh clone is ok. now I think probably git-lfs make troubles

lizhucheng, 2022-08-02:
many thanks!
the reason is I did not use git lfs correctly.
this answer works.     [https://github.com/git-lfs/git-lfs/issues/4967](https://github.com/git-lfs/git-lfs/issues/4967)

foxymiles, 2022-08-02:
Glad to hear you got it figured out!