foxglove/studio #4025: Advanced Image Annotation (SVG format)

DaniGarciaLopez, 2022-08-03:
Hi! 

Although in most situations the current state of annotations on images is sufficient, I think it would be great to be able to show more complex shapes. 

In our case we are developing a HUD video overlay and it is great to be able to toggle on/off annotations and not lose quality when zooming, but we are very limited by being able to display only points, lines and polygons.

What I suggest is the possibility to load more complex vector shapes, such as SVG, to have more freedom when creating annotations (being able to include text as well, for example). 

A possible way to implement this would be to create a new ROS message (`SVGAnnotation.msg`) with the fields: timestamp, position (origin of the SVG) and data (string with the XML information of the SVG shape). And include it in `ImageAnnotations.msg` as an array to use it together with the line and circle messages already created.

What do you think, do you have something similar in the roadmap? Any similar implementation is welcome, thanks!

defunctzombie, 2022-08-03:
> not lose quality when zooming

Do you see the annotations getting blurry when zooming? I've tried zooming with our sample data and the annotations remain crisp. If you are seeing them blurry could you share an example. Since annotation are specified using points, lines, circles as vector coordinates we are able to render them crisply at all zoom levels.

DaniGarciaLopez, 2022-08-03:
Hi @defunctzombie. Yes, the annotations are crisp. That's why we want to use annotations with more complex shapes, because our current method (overlaying the HUD on the image using opencv) produces a bitmap image that would look pixelated when zoomed in.

defunctzombie, 2022-08-03:
Adding a text annotation alongside circles and points sounds reasonable:
https://foxglove.dev/docs/studio/messages/image-annotations

The points annotation does have different modes which can produce different types of shapes with points:
https://foxglove.dev/docs/studio/messages/points-annotation-type

Is there an annotation besides text and arrays of points (which can make polygons) that would still be missing from your use case?

DaniGarciaLopez, 2022-08-03:
Creating text from circles and dots might work with straight fonts but if you want to use more rounded typography it could be a nightmare. Perhaps a solution for text would be to add a text field (as the rviz marker `visualization_msgs/Marker.msg` has) but I guess being able to import vector shapes directly makes this field unnecessary.

Let's say we want to create a simple curved line which gradually changes the width from the beginning to the end. To do this with the current method we would have to approximate the line to lots of little segments, so (depending on how many points we have defined within the line) if we zoomed in close enough we could see the actual straight segments with different width each. With SVG, on the other hand, we could simply define a bezier curve without doing any approximation. 

I also think vector shapes allow for more creative designs. Let's say we want to overlay in the image a warning signal icon when the robot runs out of battery. With the current method we would have to create the icon from scratch, using SVG we could easily create it with vector graphics editing software or download it from the Internet.

I think sending the SVG data in each message gives more freedom to create animations or reactive changes than loading it from a static file, as we do to display the 3d model from a `.dae` file in the URDF. Let's say we want to show a joystick icon that moves depending on the turn command, we would have to load many SVG files with the different position of the joystick but having the SVG definition in the message is more practical for creating seamless animations.