foxglove/studio #4032: Teleop with with joystick-like input

jasonimercer, 2022-08-03:
Would it be possible to add an advanced teleop panel that allows the control of a robot with a draggable nub? 

Options would be:
* Square limits or Circle limits. This would define the limits you could drag the nub. Square limits allows all combinations of the up-down value and the left-right value. Circle would limit max up-down based on the value of left-right and vice-versa. Releasing the nub would reset to 0.
* Dead zone sizes. These would define ranges where values from the "other" coordinate would be clamped to zero. This can make a pure turn in place easier or a pure "go forward" easier. Dead zones would show up somehow on the UI and a dead zone of 0 would remove the hint from the UI.
* "Tank" vs "Car" mode. "Tank" mode would make each coordinate decoupled. "Car" mode would negare the left-right when the up-down goes negative.
* Max/Mins on each coordinate.
* Map coordinates to linear-xyz / angular-xyz similar to your traditional teleop settings.

With the above it becomes possible to teleop a robot with finesse. The current teleop is all gas or all turns and when you factor in concepts like "about to move" low-level constraints, you get nearly uncontrollable surprises.

rgov, 2022-08-03:
It is in early alpha [here](https://github.com/ARMADAMarineRobotics/studio-extension-gamepad) and I welcome feedback on that issue tracker, but I think it's premature to release it on the marketplace just yet.

I would love to see #3413 or a comparable solution land to clear up some hacks in the code as I move forward with supporting more types of controllers (currently it bakes in the Sixaxis).

jasonimercer, 2022-08-03:
Ideally there would be a solution that could be used with a mouse and not rely on extra hardware. It is possible to walk around a facility behind a robot holding a laptop in your left arm and controlling the robot with your right hand. If there's a physical gamepad that requires 2 hands, you can no longer see what the robot sees but you still need to carry the laptop. Even if the gamepad drops down to single handed with careful balancing, you still would need to let go of the controller to click elsewhere to see extra information.

Gamepads are good input devices in some situations but are overhead on others.