foxglove/studio #4046: Add "Fixed Frame" and "Target Frame" for 3D view

PhilippSchmaelzle, 2022-08-04:
I do like the feature in rviz where you have a separate frame in which all data is transformed to --> `Fixed Frame` and also be able to set a different frame for the viewpoint --> `Target Frame`.

One example use case is attached. On the left `Fixed Frame` and `Target Frame` are the same. The car drives out of view (if not manually followed).
On the right hand side, the `Target Frame` is `base_link` and the view point moves with the frame. But all data is still visualized in the `Fixed Frame`. The 5s aggregated points are not blurred on static objects, as they do not move in map frame.

Currently I can not implement this in foxglove and would need to move the camera manually when viewing data in the map frame.

I would appreciate such a possibility :)

[image]

Best Philipp

GreatAlexander, 2022-09-02:
We would also love to have this functionality, specially since due to huge transforms and floating point issues, our target frame can sometimes be very far from the fixed frame (And it's a pain to zoom out -> scroll around to find our data -> zoom back in).

jtbandes, 2022-09-02:
I think this work might be covered under #3710 / #4329. We are adding a "stationary" mode, where the selected frame (i.e. base_link) will still be used for rendering with the best precision, but the camera position will be stationary with respect to the fixed frame (i.e. map). In our app the fixed frame is always the root-most frame of the tf tree. Does that sound like it would take care of your request?

jtbandes, 2022-09-02:
> Ideally, a Foxglove layout would be configured to "track" base_link and thus work with any ROSbag no matter where the robot is relative to map

I guess I'm confused because this is how it already works today: when you select base_link here, that means the camera will track the origin of the base_link frame. The fixed frame is still going to be "map" or whatever your root frame is (this happens automatically behind the scenes, we don't have a menu to select the fixed frame).

[image]

Perhaps your question is different from @PhilippSchmaelzle's?

PhilippSchmaelzle, 2022-09-02:
Thanks for coming back to this issue @jtbandes.
If I set `target_frame` in rviz to the same as `fixed_frame` e.g. map, one would need to search/follow manually if `base_link` is far away/moving.

For me, #4329 would not solve it completely as we would like to set a different `fixed_frame` than the root of our `tf` tree.
E.g. `tf` tree looks like this: utm -> map -> odom -> base_link

Than we would need to set `fixed_frame` to odom and the `target_frame`/camera_frame for example `base_link`.

jtbandes, 2022-09-02:
Why do you want fixed_frame to be odom? That would mean point clouds continue moving relative to the map after they are first displayed?

PhilippSchmaelzle, 2022-09-02:
As we have many different use cases where we need to display our data also in odom / map /... or other frames which are not the root of `tf`.

E.g. if we develop our localization. We need to look at the movement in odom as well as in map. And for such tasks it is helpful to set odom as `fixed_frame`.

Or imagine when you have a stationary frame (some infrastructure / road side unit) and one likes to use thats frame as `fixed_frame` which is a child of utm or map.

Therefore it is useful to be able to fixe different than just the root.

I hope I could draw a good image of some use cases.

GreatAlexander, 2022-09-06:
I've now tried #4329 with our data and can confirm it is not sufficient to solve our use case, our use case is the same as the OP (Indeed, we'd like to use 2 different frames, the `fixed_frame` and the tracked frame [as in RViz] that would control the camera's pose).

defunctzombie, 2022-10-21:
Re-visiting this so we can identify what workflow is lacking here.

> As we have many different use cases where we need to display our data also in odom / map /... or other frames which are not the root of `tf`.

The 3D panel does not display data in the root tf. It displays data in the "Display frame" you've selected.

I think it would be helpful to explain how the 3d panel in Studio renders data and how this differs from rviz `fixed frame` and `target frame` options.

In Studio when you select the `Display frame` you are selecting the coordinate frame for rendering. To render objects from different frames in the scene, there must exist a transform from the object's frame to the display frame. Studio will then calculate the pose of the object following this transform path from one frame to another. Studio is able to follow this path using the earliest common ancestor of both frames. It _does not_ have to round-trip through the root frame, it only needs to find the common ancestor of both frames. If the common ancestor is "very far" from either child frame this results in math rounding precision errors and rendering artifacts. These rendering artifacts be present in rviz as well when you select the common ancestor that is very far from the child you want to follow.

> On the right hand side, the `Target Frame` is `base_link` and the view point moves with the frame. But all data is still visualized in the `Fixed Frame`. The 5s aggregated points are not blurred on static objects, as they do not move in map frame.
> 

The objects do not move in the map but they do "move" relative to the viewport movement. How does this impact whether they are blurred?

Can you share your sample data or a video showing what blurring you experience in Studio when selecting `base_link` as the `Display frame`?

---
I want to understand the desired workflow as well as what the experience is today. As I read this thread it sounds like there might be two related but separate issues.

The first issue is around possible "blurring" within the display when following a specific frame. I'd like to see a video and ideally some sample data showing this blurring.

The second issue sounds like a request to set the common coordinate frame for transforming objects in different frames _separately_ from the frame you want the viewport to "follow". I do not _yet_ understand what specific issue this would solve because we'd still need to be able to calculate the transformations from the "follow" frame to the selected "common" frame. This is still subject to any "far" ancestor round trip math rounding.

I cannot speak to what led rviz to have a "fixed" and separate "target" frame setting but our rendering logic does not require making a distinction between these to achieve a scene render. However, I am certainly open to the idea that we've overlooked an important debugging and visualization workflow.

The best thing to help me understand is to share some sample data and describe the debugging task or workflow that you want to accomplish. You can describe this in the context of rviz and I can attempt to translate this into a Studio workflow or you can try using Studio and report where it is not allowing you to continue debugging your issue.

jhurliman, 2022-10-23:
> Studio is able to follow this path using the earliest common ancestor of both frames. It does not have to round-trip through the root frame, it only needs to find the common ancestor of both frames.

This is not correct. Studio uses the concept of a "root frame" or "fixed frame" and round-trips all pose transformations through it. This frame is automatically set to the top-level (root) frame relative to the current render frame (display frame). We've talked about allowing the user to choose the fixed frame like rviz does, but it's not implemented today.

defunctzombie, 2022-10-23:
> This is not correct. Studio uses the concept of a "root frame" or "fixed frame" and round-trips all pose transformations through it. This frame is automatically set to the top-level (root) frame [...]

When did this change? I thought we had the common ancestor behavior in the legacy 3d panel as a means of avoiding round trips through far frames by using static "local" frames. Do we not have that ability?

If I have `map -> local static -> base_link -> arm` and `map -> local static -> object` and am rendering in `local_static` we will round trip lookups of `base_link` and `object` through `map`? Seems like extra hops for no reason?

If this was always the case then I am mistaken on this. Why do we not pick the common ancestor for the transformation? We don't actually need a "root frame" for performing the transformation given how we render the scene around the display frame.

defunctzombie, 2022-10-23:
Took a look at the code and confirmed we use a _root_ frame in the legacy 3d panel as well. I had incorrectly thought we changed this at some point to use a common ancestor because we center rendering around the _display frame_. Seems we do not.

Given that learning I see why there's a request to set the common coordinate frame. I thought we were already doing the smart thing here with common ancestor detection but it seems we are not.

PhilippSchmaelzle, 2022-10-24:
thanks @defunctzombie for revisiting this issue. 
I try to re-frame the issue I see with Studio at the moment as #4329 added some improvements and I can't fully recreate the problem I had at the beginning with it. <-- my initial question keeps some open questions...

> Took a look at the code and confirmed we use a _root_ frame in the legacy 3d panel as well. I had incorrectly thought we changed this at some point to use a common ancestor because we center [...]

I think this might be the right step. The following *gif should show the difference of `rviz`vs `Studio`. 
**What do you see:**
  [Data decay 5s]
*left hand side:*
- rviz
  - left: `fixed_frame: map` - `target_frame: base_link`
  - right: `fixed_frame: base_link` - `target_frame: base_link`
*right hand side:*
- Studio
  - top left: `Display frame: map` - `Follow mode: Position`
  - bottom left: `Display frame: odom` - `Follow mode: Position`
  - right: `Display frame: base_link` - `Follow mode: Position`
Within the *rqt_reconfigure* I publish a new transform between `map --> odom`. 

https://user-images.githubusercontent.com/64705964/197545111-a4503d74-1e0a-4b19-9341-3a519fcb6746.mp4

In rviz, this is only visible on the left as the data is rendered in `map` and the sensor data has to pass the `odom -> map` transform. But on the right rviz view there is no impact as the rendering frame is `base_link` and from the LiDAR to `base_link` is only a static transform.

On the other side in *Studio* this jumping is visible in all three views. For me this looks like the data visualized is always in `root (map)` frame. 

A problem I see - for our workflow - with this method is that it is not clear which transform influences the result. E.g. localization: There is one Kalman-Filter working on the tf of `odom->base_link` and one KF working on the tf of `map->odom`. Now I want to see how well performs the one and the other. Visually this can be done by looking at the LiDAR data in the related frame (`map` or `odom`) and how well it represents static objects as such (if for example a static wall shows as a solid/clear edged line or is it blurred). If the wall is now blurred in `odom` I know, that `odom -> base_link` is not good. And if it is blurred in `map` but not in `odom` I know that `map -> odom` has to be improved. 
But in Studio it looks like this method is not possible as the jump in `map -> odom` is visible in all three settings.

The next video shows what happens if `base_link` starts moving.
**What do you see:**
 [Data decay 5s]
*left hand side:*
- rviz
  - left: `fixed_frame: odom` - `target_frame: base_link`
  - right: `fixed_frame: base_link` - `target_frame: base_link`
*right hand side:*
- Studio
  - top left: `Display frame: map` - `Follow mode: Position`
  - bottom left: `Display frame: odom` - `Follow mode: Position`
  - right: `Display frame: base_link` - `Follow mode: Position`

https://user-images.githubusercontent.com/64705964/197544306-f27104c8-94b4-4fbc-ab36-8fa66ca06212.mp4

In rviz, the left window shows static objects clear and with sharp edges. Also when `base_link` starts moving as `odom` is a world fixed frame. The right window blurs out and you can't see anything. But that's what I would expect if I look on 5s LiDAR data which moves and the tf from LiDAR to `base_link` is only static.

In Studio nothing blurs as everything is in `map?`. An improvement is the the Follow feature which works fine for the right panel. In the left panel it moves out of sight as the display frame is a fixed one - makes sense was well. 

**Conclusion:**
What I now think would be super helpful is:
- do not transform all data to `root` as it disables some sort of debugging. But take the first common parent.
- I would also say it reflects a wrong impression of the data. As I think the intuition would be, that visualized data in `odom` only went from `SENSOR -> base_link -> odom` and not through `map/root`.
> If I have map -> local static -> base_link -> arm and map -> local static -> object and am rendering in local_static we will round trip lookups of base_link and object through map? Seems like extra hops for no reason?

Do you truly go back up from `map` to `local_static` or do you stop in `map`? because I would think that the jump I induced in the first video should then only be visible in the map window (top left) as `odom -> map -> odom` should equal itself out. But maybe I am wrong here.

- Provide a option to set a *camera_frame* or so. Such that the user can set a frame in which the camera is in (rviz = target_frame) and a frame in which the data is represented in (rviz = fixed_frame)

defunctzombie, 2022-11-03:
@banisadr Putting this on your radar as well. We should be able to identify the gap in workflow and capability that we have and come up with a proposal for addressing that gap and allowing for the type of debugging that needs to happen.