foxglove/studio #4057: Initial messages are sometimes skipped when looping with new player

jhurliman, 2022-08-05:
**Description**

I ran into this issue while troubleshooting another unrelated bug report. In the attached mcap, there are 101 evenly spaced (in time) `/tf` messages and 3 `/cubelist` messages at t=0.01, t=2, and t=4. While playing back with looping enabled, it seems like the initial message is often missed (my layout has one New3D panel and one Raw Message panel).

[filled-empty-filled.mcap.zip](https://github.com/foxglove/studio/files/9264872/filled-empty-filled.mcap.zip)

defunctzombie, 2022-08-05:
The initial message on which topic? The `/cubelist` topic or `/tf` ?

jhurliman, 2022-08-05:
@defunctzombie the initial topic on `/cubelist`. It is skipped in both raw messages and New3D so I don't think this is a panel issue, it appears to never come over the message pipeline.

defunctzombie, 2022-08-06:
Confirming I see this behavior when looking. The initial playback is good but on loop the first message is missing.

PR #4074 (2022-08-09): Wait for previous emit state in tick

**User-Facing Changes**
Fixes bugs in the latching player when using a slow framerate or looping.

**Description**
In the iterable player, when the previous message is still past the target tick time, the logic would override the messages and current time without waiting on the current emitState to finish. This would result in overriding outgoing messages.

This change fixes this behavior to wait on any active emit state to finish before proceeding to emit state for the new message data.

Fixes: #4073
Fixes: #4057
commit b3047cc6 (2022-08-09): Wait for previous emit state in tick (#4074)

In the iterable player, when the previous message is still past the target tick time, the logic would override the messages and current time without waiting on the current emitState to finish. This would result in overriding outgoing messages.

This change fixes this behavior to wait on any active emit state to finish before proceeding to emit state for the new message data.

Fixes: #4073
Fixes: #4057
closed 2022-08-09 by defunctzombie