foxglove/studio #4059: Timestamp is not correct when playing .mcap file

felixf4xu, 2022-08-05:
**Description**

Hi,

I have created a .mcap file and I can play it in Foxglove Studio, now I'm trying to use the next/previous button to play the data frame by frame.

There seems to be a timestamp indicating current frame's time, but if I click the previous/next button back and forth, the timestamp is changing every time, it's not reading from my original frame data's timestamp.

- Version:
- OS:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**
[image]

1. remember current timestamp,
2. click the previous button once, then the next button once, 
3. check the timestamp if equal to the timestamp in step 1.

**Expected Behavior**

the timestamps are same, and it's from the timestamp within the .mcap file.

**Actual Behavior**

the timestamps are not same, and it seems to be random.

defunctzombie, 2022-08-05:
Duplicate of: https://github.com/foxglove/studio/issues/3872

I am working on a fix here: https://github.com/foxglove/studio/pull/4045

jtbandes, 2022-08-05:
Duplicate of https://github.com/foxglove/studio/issues/3872

PR #4045 (2022-08-14): Add optional playUntil method to Player interface
**User-Facing Changes**
Forward seek and backwards seek step forward and backwards the same amount of time.

**Description**
Our current behavior for forward seeking (via arrow key or seek button), is to "play" the data forward at least the seek amount. The purpose of this behavior is to ensure replay of stateful messages (like tf and markers) so pressing "seek" from the start of the file results in the same playback as playing from the start of the file.

Unfortunately, the way we implemented this feature was to start playback and attempt to p [... cut]
commit b291063e (2022-08-14): Add optional playUntil method to Player interface (#4045)

**User-Facing Changes**
Forward seek and backwards seek step forward and backwards the same amount of time.

**Description**
Our current behavior for forward seeking (via arrow key or seek button), is to "play" the data forward at least the seek amount. The purpose of this behavior is to ensure replay of stateful messages (like tf and markers) so pressing "seek" from the start of the file results in the same playback as playing from the start of the file.

Unfortunately, the way we implemented this feature was to start playback and att [... cut]
closed 2022-08-14 by defunctzombie