foxglove/studio #4065: onChange event fires twice with enter key in VisibilityToggle

DoParkEQ, 2022-08-06:
**Description**
There is a keyboard interaction bug from Material UI's Button component, which triggers the `onClick` event when `onKeyDown` event is fired by pressing the enter key. This bug is initially found from `VisibilityToggle`.

Version: v1.20.1
OS: macOS 11.4

**Steps To Reproduce**

[image]

**Expected Behavior**
Pressing enter key should change its visibility state. 

**Actual Behavior**
When the enter key is pressed, the onChange event fires twice therefore, the visibility state remains the same.

jtbandes, 2022-08-06:
Thanks for the report. Would you be willing to create a PR to fix this issue?

This 3D panel is in maintenance mode, we would recommend you use the 3D (Beta) panel.

DoParkEQ, 2022-08-07:
@jtbandes for sure! I am happy to contribute.

PR #4067 (2022-08-08): Visibility toggle button bug fix 
**User-Facing Changes**
Fixed an issue in the 3D panel where toggling topic visibility with a keyboard did not work.

**Description**
MUI's Button component presents a bug that triggers `onClick` event when `onKeyDown` event fires. `onChange()` has a conditional statement that checks whether the event is automatically populated.  

Fixes #4065
commit 3f1d72e4 (2022-08-08): visibilityToggle - filters out automated onClick event when enter key is pressed (#4067)

**User-Facing Changes**
Fixed an issue in the 3D panel where toggling topic visibility with a keyboard did not work.

**Description**
MUI's Button component presents a bug that triggers `onClick` event when `onKeyDown` event fires. `onChange()` has a conditional statement that checks whether the event is automatically populated.  

Fixes #4065
closed 2022-08-08 by jtbandes