foxglove/studio #4068: latching feature slow down image panel

lizhucheng, 2022-08-08:
**Description**
when latching feature is on, image panel  block at some moment.

- Version: 1.20.1
- OS: ubantu 20.04 focal （8core 16G）or a 8 cores 16G MacBook Pro os 12.2.1
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): a rosbag file with 400M in one minute record data. a rosbag within some radar or lidar data will match it. (I'm sorry I can not post a real rosbag for some security reasons.
)
- ROS distro (if applicable): null

**Steps To Reproduce**

1. a rosbag within a Video;
2. layout includes 3d and image pannel and a raw message panel;
platform: ubantu 20.04 focal （8core 16G） 

**Current Experience**
image panel will be slow. (video blocks at some moment)

**Desired Experience**
video plays fluently

defunctzombie, 2022-08-09:
@lizhucheng Without a bag file there's not enough information here for us to reproduce this issue. There's far too much variance in what could be happening from how fast you are trying to play back, how much data exists at each frame, how your bag file is structured.

Feel free to message me privately on our slack (or email me) if you want to share the file privately and we'd be happy to investigate once we have data to reproduce. If it helps - you can blur or make mock data - we just need something that reproduces the issue.

defunctzombie, 2022-08-09:
Possibly related issue: https://github.com/foxglove/studio/issues/4069 - we've also requested a bag file to reproduce.

closed 2022-08-09 by defunctzombie

lizhucheng, 2022-09-13:
> @lizhucheng Without a bag file there's not enough information here for us to reproduce this issue. There's far too much variance in what could be happening from how fast you are trying to play back, [...]

I got a issue data. Can you post me your email address?

defunctzombie, 2022-09-13:
@lizhucheng I sent you an email.

lizhucheng, 2022-09-13:
got it