foxglove/studio #4069: Image pane is very slow

felixf4xu, 2022-08-08:
**Description**
Hi,

I'm using `foxglove/RawImage` and websockets to display some online image data, the encoding is `bgra8`. the image size is 1920x1080.
```c++
rawImage.set_encoding("bgra8");
```
the current result is that the image display is very lagging, it seems that the images are buffered.

I opened another pane of `RawMessage` to see the protobuf data received without delay, so I assumed that the data is _received_, but the display/rendering is slow. 

I just hoped that I can confirmed that the ecoding of `bgra8` itself is slow, not the foxglove display.

I can do some experiment as mentioned in the folloing link, try jpeg with lower resolution, but in my situation, ideally the format conversion/resizing happens on my PC (where foxglove is running) instead of my embedded device with a camera so that I save some cpu power on the embedded device.

>960x604 JPEG images (quality level 85) at 30Hz

https://github.com/foxglove/studio/issues/3669#issuecomment-1163727790

- Version:
- OS: Windows
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): websockets, protobuf

jtbandes, 2022-08-08:
Can you record the data into an mcap file and share it with us so we can reproduce the performance issues?

felixf4xu, 2022-08-09:
do you have some tool to record websocket protobuf messages to mcap file? otherwise I need to write some code to convert it to mcap file

jtbandes, 2022-08-09:
Sorry we don’t have a simple tool for it. There is some example code here: https://github.com/foxglove/mcap/blob/main/cpp/examples/protobuf/writer.cpp

If you can share the code for the websocket server, that works too.

felixf4xu, 2022-08-12:
I was wrong in the previous statement, the message was not received in the expected frequency.

If I test on the same PC with `RawImage`, the display fps will increase a lot;
if I test with `CompressedImage`, the display fps can match the data sending frequency.

So it's more like a data transfer speed issue, not a  foxglove display issue.

closed 2022-08-12 by felixf4xu