foxglove/studio #4072: Latching data platform player

amacneil, 2022-08-08:
**Description**

I'm looking at [this URL](https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_maadgdhBdV3xMDHi&ds.end=2018-08-01T19%3A56%3A02.548000000Z&ds.start=2018-08-01T19%3A55%3A43.298570000Z&layoutId=822c89a6-98b4-4fca-a1fa-ef4b76b55f22&time=2018-08-01T19%3A55%3A50.124423859Z) from data platform demo org.

For some reason, the map layer disappears from both old + new 3d panels when using latching player. The messages are showing fine in raw messages panel. Disabling latching player fixes the bug.

https://user-images.githubusercontent.com/637671/183517154-ccdb899e-644d-4f5e-b3cb-8bfcef2e7d6a.mov

- Version: main
- OS: web / macos
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): data platform

defunctzombie, 2022-08-09:
@amacneil is that layout linked in the request a team layout? I'm getting an error saying the layout cannot be found. Which org is this?

amacneil, 2022-08-09:
It is just a personal layout. You can create one similar to what I had - all you have to do is add a 3d panel and enable some topics.

This is in the "james" demo org.

defunctzombie, 2022-08-09:
Few things we've learned here:

1. The request for "lookback" on the `/tf` topic is not returning the `/tf` message at the start of the time range (it is returning one prior to the start) even tho we know there is one at the start.
2. There is some issue with playing from the start that doesn't include the `/tf` message that should be at the start.

A few related console and data platform tickets have come from this:
- https://github.com/foxglove/console/issues/946
- https://github.com/foxglove/console/issues/947

Part of the issue here seems to be that data lookback is not including the specific start time so we are ending up with a `/tf` frame that isn't the frame we want.

There's an additional issue here around why playback is not playing the next expected `/tf` message but I'd like to look into that once we've resolved the API issues.

defunctzombie, 2022-08-16:
This PR https://github.com/foxglove/studio/pull/4144 might address some parts of this. Some of this is specific to how our sample data is timestamped.

defunctzombie, 2022-08-17:
@amacneil are you still able to reproduce this?

amacneil, 2022-08-17:
Yes, still happens for me [here](https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_maadgdhBdV3xMDHi&ds.end=2018-08-01T19%3A56%3A02.548000000Z&ds.start=2018-08-01T19%3A55%3A43.298570000Z&layoutId=f41606d9-1ec2-4b44-b471-f19687c2a742&time=2018-08-01T19%3A55%3A43.330046654Z).

defunctzombie, 2022-09-02:
@amacneil We've landed the api and streaming fixes. I ran through this again and could no longer reproduce this issue. I am going to close this as fixed given the underlying fixes we landed. If you are still able to reproduce please re-open with what else I should look for when playing back the data.

closed 2022-09-02 by defunctzombie