foxglove/studio #4082: Map layer opacity affects objects in front of it

amacneil, 2022-08-10:
**Description**
When I adjust the opacity of my map layer, it affects the visibility of objects in front of the map.

https://user-images.githubusercontent.com/637671/183789037-f47c198c-e8d2-4133-a862-9b3f88c79372.mov

- Version: main
- OS: web / macos
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): sample nuscenes

amacneil, 2022-08-18:
Reopening because we still have some unresolved issues. On certain angles, markers end up below the map in our sample dataset, making them near invisible.

https://user-images.githubusercontent.com/637671/185268542-06ca35cc-206f-4665-9146-8053b3c92e30.mov

@jtbandes pointed out this only happens with transparent objects. I was able to confirm this with a user script rewriting our transparent markers to be solid. I think it only occurs when two transparent objects intersect: making either object solid seems to fix it. Unfortunately our sample dataset heavily uses transparent objects.

```ts
import { Input, Message } from "./types";

export const inputs = ["/markers/annotations"];
export const output = "/markers/solid";

export default function node(event: Input<"/markers/annotations">): Message<"visualization_msgs/MarkerArray"> {
  const m = event.message;

  m.markers.forEach((n, i) => {
    m.markers[i].color.a = 1;
  });
  return m;
}
```

jhurliman, 2022-08-22:
This is the expected result of trying to render many translucent objects of varying sizes close to each other from arbitrary camera angles. Games traditionally get around this limitation by not doing that. More advanced renderers will use some form of Order Independent Transparency, which I've broken off into a separate ticket at #4320. To improve the experience of our sample data, I'm looking at scene tweaks we can make to avoid the popping and strange alpha blending.

PR #4146 (2022-08-17): render order being disabled causing transparency fighting
Fixes #4082 - Enable threejs object sorting

**User-Facing Changes**
 - Transparent object ordering in the new 3D view has been fixed so that their colors can be properly stacked in order away from the camera
 
**Description**
 - enable sortObjects on the threejs renderer
 - doesn't seem to affect the render order or lines and text markers/labels

https://github.com/foxglove/studio/issues/4082
commit 9133d651 (2022-08-17): render order being disabled causing transparency fighting (#4146)

Fixes #4082 - Enable threejs object sorting
User-Facing Changes

Transparent object ordering in the new 3D view has been fixed so that their colors can be properly stacked in order away from the camera

Description

enable sortObjects on the threejs renderer
doesn't seem to affect the render order or lines and text markers/labels

#4082
closed 2022-08-22 by jhurliman