foxglove/studio #4083: User Script causes Image panel to stop updating

jtbandes, 2022-08-10:
**Description**
When a user script is added, the image panel does not update as expected.

- Version: latest web
- OS: Chrome / macos

**Steps To Reproduce**
- Create a layout with one Image panel
- Load https://librealsense.intel.com/rs-tests/TestData/stairs.bag
- Select the `/device_0/sensor_1/Color_0/image/data` topic
- Play the bag and observe that images appear and continue to update through the end of the bag.
- Add a User Scripts panel and copy/paste the user script from #4080.
- Load the bag again and play. Observe that a black frame appears, but then subsequent frames do not appear.

https://user-images.githubusercontent.com/14237/183790033-1c0732af-ce38-40c1-a71e-e7c2cb7d24ef.mov

snosenzo, 2022-08-19:
interact or changing another panel causes the image panel to be fixed
changed topic in raw message panel from undefined to camera_info
 - appears to work for any topics even those not related to device_0
changing the layout (width/height of any panel) fixes the image panel

makes me suspect that the image panel is missing some kind of update/rerender that should happen when the file is read in. I've been fishing around but can't seem to find a good place/file to start -- any ideas?

jtbandes, 2022-08-19:
I believe the user script code has an error (see also https://github.com/foxglove/studio/pull/4178) so the question is why that error interferes with other topics. All the management of subscriptions and injecting the user script messages happens in the UserNodePlayer so I'd start looking there (maybe after the problem starts happening, set a breakpoint to see when the next image message is read from the underlying player, and where it ends up going).

snosenzo, 2022-08-22:
There doesn't appear to be an error in the user script from what I can tell. Also after some further testing it looks like this happens with the image panel even without the user script being present. All you have to do is have the image panel up and reading from the bag, then add the same data source again, and the same result happens where the image is stuck. It looks like there's some difference from the instantiation on startup vs the instantiation post-startup from the data source.

snosenzo, 2022-08-23:
I've tabled this for now since I can't seem to pinpoint where something is going wrong. I've looked at `_getMessages` in the `UserNodePlayer` and found that it's seems to be acting properly. The problem seems to be that `_onPlayerState` function is being called with empty message arrays where there should be new messages. So whatever is responsible for populating the messages in the `activeState` is not doing so after the first frame.

commit ea8b36ea (2022-08-25): User Script causes Image panel to stop updating
Fixes #4083
PR #4277 (2022-08-25): Fixes setPlaybackSpeed from Overriding Backfill in Iterable Player
**User-Facing Changes**
 - not worth mentioning?

**Description**
 - setPlaybackSpeed was setting state back to idle before the playback iterator can be initialized
 - because the playback iterator wasn't initialized messages weren't getting added to state, thus causing the panels to not receive the new messages

Fixes #4083
commit b2efcfae (2022-08-25): Fixes setPlaybackSpeed from Overriding Backfill in Iterable Player (#4277)

**User-Facing Changes**
 - not worth mentioning?

**Description**
 - setPlaybackSpeed was setting state back to idle before the playback iterator can be initialized
 - because the playback iterator wasn't initialized messages weren't getting added to state, thus causing the panels to not receive the new messages

Fixes #4083
closed 2022-08-25 by snosenzo