foxglove/studio #4084: "User Script did not produce a message" but no error details

jtbandes, 2022-08-10:
**Description**

- Version: latest web
- OS: Chrome / macos
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): https://librealsense.intel.com/rs-tests/TestData/stairs.bag

**Steps To Reproduce**
- Load this data source: https://librealsense.intel.com/rs-tests/TestData/stairs.bag
- Add a user script with this code:
```ts
import { Input, Message } from "./types";
type Output = Message<"foxglove_msgs/ImageAnnotations">;
type PointsAnnotationMsgType = Message<"foxglove_msgs/PointsAnnotation">;

export const inputs = ["/device_0/sensor_1/Color_0/image/data"];
export const output = "/studio_script/point_annotations";

export default function node(
  event: Input<"/device_0/sensor_1/Color_0/image/data">
): Output {
  const out = {} as Output;
  out.points = new Array();
  const line_points = {} as PointsAnnotationMsgType;
  line_points.points = new Array();
  line_points.outline_colors = new Array();
  (line_points as any).outline_color = { r: 1, g: 1, b: 1, a: 1 };
  let x0: number = 0;
  let x1: number = 1920;
  let y0: number = 500;
  let step: number = 10;
  let i: number = 0;
  while (x0 + i * step <= x1) {
    let xi: number = i * step;
    line_points.points.push({ x: xi, y: y0 });
    i++;
  }
  line_points.type = 3;
  line_points.timestamp = event.message.header.stamp;
  if (line_points.points.length % 2 !== 0) {
    line_points.points.length -= 1;
    line_points.outline_colors.length -= 1;
  }
  out.points.push(line_points);
  return out;
}
```

**Expected Behavior**
Some error should show me why my user script didn't produce a message.

**Actual Behavior**
[image]

jtbandes, 2022-08-17:
You also have to enable the annotation topic in an Image panel:

[image]

defunctzombie, 2022-08-17:
[image]

When I open the user script editor I see this problem.

So there is an additional error shown.

PR #4178 (2022-08-18): Improve error message when user script errors during runtime

**User-Facing Changes**
When a user script errors, the new message displayed to the user is the following:

[image]

Additionally, the `Problems` tab diagnostics list is fixed to removed the `undefined` text that would show up after `Runtime`.

[image]

**Description**
When a user script errors during runtime and fails to produce a message, the previous error displayed would only indicate that a script failed to produce a message. This warning did not make it obvious that the user script in fact errored during runtime.

This change  [... cut]
commit 3e16e267 (2022-08-18): Improve error message when user script errors during runtime (#4178)

When a user script errors during runtime and fails to produce a message, the previous error displayed would only indicate that a script failed to produce a message. This warning did not make it obvious that the user script in fact errored during runtime.
This change improves the error to be clear that the script errored during runtime when there is an error rather than only a missing message.

Fixes: #4084
closed 2022-08-18 by defunctzombie