foxglove/studio #4088: Playhead is not at the beginning when loading a bag

jtbandes, 2022-08-10:
**Description**
The playhead appears slightly offset from the beginning of the bag when loading a bag, although the player starts paused and no data is displayed.

[image]

- Version: latest web
- OS: Chrome macOS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): https://librealsense.intel.com/rs-tests/TestData/stairs.bag

jtbandes, 2022-08-12:
Downgrading to UX. Sounds like the time displayed on the playhead is accurate.

defunctzombie, 2022-09-06:
This behavior is by design. The intent of this feature is to make an effort to read some messages from the file so the various panels in the UI display some data rather than sitting empty and waiting for the user to start playing before any data is loaded since rarely are all the desired messages at the exact start time of the file (usually only one or two are exactly at the start).

I'm going to close this as not planned. I'll keep an eye out for any user's that surface this as confusing behavior and if we see that we can consider revisiting.

closed 2022-09-06 by defunctzombie