foxglove/studio #4089: [3d beta] Option to override opacity and/or color for markers

amacneil, 2022-08-10:
My file (nuscenes example) has opacity set in the markers. I am trying to view the file and decided I don't like the opacity that the message author selected, so I would like to override the opacity. I think it would also be nice to override the colors. 

Personally I would prefer to do this for all namespaces in the marker topic, but I would settle for having it configurable per-namespace if that is easier from a settings UI perspective. 

The old 3D panel allowed overriding colors (but not opacity), so this is a partial regression in functionality.

[image]

jtbandes, 2022-08-10:
Duplicate of https://github.com/foxglove/studio/issues/3943? Or are these both halves of the same issue?

amacneil, 2022-08-10:
Oh yeah I guess I already complained about this :)

I do think it could be useful to override opacity only (rather than both color+opacity), for marker topics that already have colors (like our nuscenes demo). But I will close this and hold my judgement until after https://github.com/foxglove/studio/pull/4054 lands.

closed 2022-08-10 by amacneil