foxglove/studio #4091: Latching player: Sometimes some topics get "stuck" while others keep playing

amacneil, 2022-08-10:
**Description**

- Version: main
- OS: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): sample nuscenes

Sometimes during playback, my map and markers freeze up, but the lidar keeps playing.

https://user-images.githubusercontent.com/637671/183826194-7562a6fe-ecd9-4636-92c7-c00c8a5d731f.mov

amacneil, 2022-08-10:
Here is a better video with some raw message panels shown:

https://user-images.githubusercontent.com/637671/183827313-0e57ff3c-541f-48cd-914f-f34c2ac74a9e.mov

There doesn't seem to be any obvious reason why this starts happening, but it does "fix" itself when the new tf message arrives.

Note this was on https://github.com/foxglove/studio/pull/4087 branch.

defunctzombie, 2022-08-12:
I can confirm I am also seeing a behavior here that looks like missing messages. What I am seeing is that during playback, I can seek forward up-to and past a time when I know there is a _/tf_ message yet I won't see this message in the raw message panel.

There should be messages at:

* 1532402929.697797000
* 1532402930.152601000
* 1532402930.648325000

But I can definitely play forward in a way that causes one of these to be "skipped" which makes the scene look "stuck" because the /tfs are old. The annotations are in _map_ frame but there's no update of _map_ to _base-link_ so the annotations follow along with base link when instead they should appear further behind.

amacneil, 2022-08-18:
This is still happening for me on latest main.

defunctzombie, 2022-08-18:
@amacneil I am unable to reproduce this issue on studio.foxglove.dev. What topics are getting "stuck" for you? The behavior in the original video is no longer happening for me.

amacneil, 2022-08-18:
Sample nuscenes, fresh layout with only 3D beta panel, all topics enabled.

Always, the lidar topic keeps moving, and every other topic gets stuck:

https://user-images.githubusercontent.com/637671/185271017-9a75629d-d4d8-4990-90dd-c89cbbe600dc.mov

Lmk how I can help debug.

defunctzombie, 2022-08-18:
Looking closer at the video - what I see is that the car has stopped moving forward relative to the map. The lidar looks like it is still "moving" but what I see is that the lidar is getting updated with new messages and the points of the lidar are moving. The lidar itself is stationary relative to the car and other scene entities. I suspect that the marker messages are also still updating but are too stationary. These items would be stationary if the `map -> base_link` transform is not present or updating. That would result in the car no longer moving "forward" in the scene and all the markers and sensor readings looking "stuck" because their frames are all relative to base_link.

defunctzombie, 2022-08-18:
@amacneil if you can make a recording of the above video with a raw message panel alongside the 3d panel. Subscribe to the `/tf.transforms[:]{header.frame_id=="map"}` in the raw message panel and lets see if that produces anything more to help debug.

amacneil, 2022-08-18:
https://user-images.githubusercontent.com/637671/185301021-86b7685d-d3ba-4ede-8f80-d3ac042e5b35.mov

https://user-images.githubusercontent.com/637671/185301057-577388a1-fe0b-4e6c-b74f-57522fb88de8.mov

https://user-images.githubusercontent.com/637671/185301242-252703d2-f971-433c-9608-e2bc4d164c53.mov

amacneil, 2022-08-19:
https://user-images.githubusercontent.com/637671/185718451-3f5da59e-16a2-4dc4-8c1a-ff18374cb5da.mov

jtbandes, 2022-08-20:
Latest discovery from the above debugging session is that PanelExtensionAdapter overwrites its `latestPipelineContextRef.current` and can drop frames if the requestAnimationFrame does not happen by the time the previous pipeline context is processed. This could be remedied by adding correct usage of `pauseFrame` to the PanelExtensionAdapter.

PR #4126 (2022-08-15): Fix missing messages from CachingIterableSource
**User-Facing Changes**
Fixes messages drops when playing data from data sources which contain multiple messages at the exact same log time (or receive time).

**Description**
When reading from the cache, the CachingIterableSource uses a binary search to deterline where to start reading in a cache block. If a block contained multiple messages at the exact same time as the search parameter, the search algorithm would return a valid index for any time matching the search parameter not necessarily the _first_ occurrence in the block. By not returnin [... cut]
commit bae47136 (2022-08-15): Fix missing messages from CachingIterableSource (#4126)

**User-Facing Changes**
Fixes messages drops when playing data from data sources which contain multiple messages at the exact same log time (or receive time).

**Description**
When reading from the cache, the CachingIterableSource uses a binary search to deterline where to start reading in a cache block. If a block contained multiple messages at the exact same time as the search parameter, the search algorithm would return a valid index for any time matching the search parameter not necessarily the first occurrence in the block. By not r [... cut]
PR #4233 (2022-08-25): Call pauseFrame/resumeFrame in PanelExtensionAdapter
**User-Facing Changes**
Extension panels wait to receive new frames until they are done processing their existing frames resulting in fewer message drops for extension panels when the computer is missing animation frames.

**Description**
When handling context updates it is important to call pauseFrame to indicate that the panel will perform some work to render the frame and resumeFrame when the panel's work is complete. Without these calls, the message pipeline will assume the panel has completed the frame and will feed another frame to the [... cut]
commit 917f44e9 (2022-08-25): Call pauseFrame/resumeFrame in PanelExtensionAdapter (#4233)

When handling context updates it is important to call pauseFrame to indicate that the panel will perform some work to render the frame and resumeFrame when the panel's work is complete. Without these calls, the message pipeline will assume the panel has completed the frame and will feed another frame to the panel to render. This additional frame will override the previous and result in dropped messages for the panel.

This change adds pauseFrame/resumeFrame logic to the PanelExtensionAdapter. As part of this change I've re-organized [... cut]
closed 2022-08-25 by defunctzombie