foxglove/studio #4093: 3D takes ALL available tf2_msgs/TFMessage into account not only from /tf

PhilippSchmaelzle, 2022-08-10:
**Description**
When I replay / re-simulate a bag file it occurs that the *tf tree* (static and dynamic) has to be recalculated. 
Therefore I use `rosbag play ... /tf:=/trash0 /tf_static:=/trash1 ...` and then start my nodes which publish the new calculated `/tf and /tf_static`.

Foxglove studio 3D visualization (only this was tested) gets now messed up as it lists all available transformation. Not only those on the `/tf` and `/tf_static` topics. 

As there are now two sources for the same transform 1.) `/trash0/1` from the bag replay and 2.) `/tf /tf_static` from the re-simulation it can happen that the visualization jumps wildly as the pose for the same *chield - parent* relation can be different.

- Version: Foxglove Studio version 1.20.1
- OS: Ubuntu 20.04
- Data source: bag file, ROS 1 native
- ROS distro: Noetic

**Steps To Reproduce**

Take a bag with `/tf` which holds some transformations e.g. map->base_link. Replay it with a remapping of `/tf:=/something` and use an `tf_broadcaster` which publishes a slightly different map-base_link transform.

The links should now jump in between both of them. 

I attached a video as well as some screenshots.

**Expected Behavior**
I would expect that the visualization subscribes only to `/tf` and `/tf_static` topics to receive all the TF-Tree information. As I think this is the standard. Otherwise working with tf-tree re-simulation is only possible if the original `/tf` topic is deleted from the bag completely.

**Actual Behavior**

In this example I remapped `/tf` to the topic name `/w13`. In the transform view of 3D panel one can see that the pose jumps between the one of tf and w13.

https://user-images.githubusercontent.com/64705964/183942551-e0c5f01f-bf6f-4699-9595-98d86d08db2c.mp4

[image]

[image]

jtbandes, 2022-08-10:
This is behaving as intended. If this behavior is problematic, we may need to introduce user settings to override it.

PhilippSchmaelzle, 2022-08-10:
Thx for your fast reply!
I did not know that this should be the behavior. At least for us it would be better if there would be an user setting.
e.g. default this behavior and a checkbox [ ] `use only tf tf_static` or so?

To migrate to foxglove this use case is mandatory at least for us.  
Appreciated

PhilippSchmaelzle, 2022-10-24:
Are there some updates on this and if you plan to integrate such?

jtbandes, 2022-10-26:
cc @banisadr for thoughts on priority