foxglove/studio #4097: Local files should show as fully buffered in scrub bar

amacneil, 2022-08-10:
**Description**
For remote files, our seek/scrub bar shows me which parts of the file have been downloaded. That is a helpful feature.

For local files, our seek/scrub bar shows me which parts of the file have been read into memory. This is not useful information for typical workflows, and I cannot think of any other tools that show me which parts of a local file have been read into memory.

https://user-images.githubusercontent.com/637671/183996811-d9e12c1e-0526-46dc-ba97-87a4bdb054bc.mov

We should change this to always show local files as fully buffered.

- Version: main
- OS: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): local bag/mcap file

defunctzombie, 2022-11-04:
> This is not useful information for typical workflows,

It impacts how quickly you can seek in an area within Studio. If an area is buffered into memory you will be able to seek faster than if you jump to an area with no buffering and we need to load and decompress your chunks.

amacneil, 2022-11-04:
Agreed, given our current performance and constraints I think this is useful information to surface to the user.

closed 2022-11-04 by amacneil