foxglove/studio #4100: After switching org, studio should select first available personal layout

amacneil, 2022-08-10:
**Description**
After switching org, Studio forgets that we have an active layout.

We should select the first personal layout (if there is one) if the app starts with no layout and personal layouts are available.

[image]

- Version: main
- OS: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): data platform

snosenzo, 2022-08-18:
Looks like it's updating the url with a proper last layout for the new org that you've switched to, but it's not reflecting that in the page itself. So an easy way to reproduce this issue is to just have an invalid layoutId in the url and load the page.

snosenzo, 2022-08-18:
oh actually, it looks like the `PlayerManager` component is specifically updating the URL with the layout whose name matches the `sourceName` of the sample data. so it doesn't update the url from whats input in any other case. looks like just setting the selectedLayoutId is only updating the URL though and not propagating through to the actual layout

snosenzo, 2022-08-23:
So @foxymiles and I have discussed that this might not be worth fixing right now with a less wholistic solution given that past changes to initializing state from the url have broken things in production and that this seems like a fairly small corner case that doesn't have a very big impact currently. I do have a working solution (from what I can tell, just need to fix tests) on the related PR, but it will probably only be temporary as there will be a greater effort to re-architect handling of organizations across the suite. Let me know if this is important and I should continue working on it given some of the risks