foxglove/studio #4107: Chrome studio - Rosbridge to remote IP not working

Chick92, 2022-08-11:
Hi, Fantastic piece of software!

My use case requires running Foxglove studio on an android tablet, i.e using the chrome extension. 

I've installed rosbridge server, and configured the launch file with the correct IP address, but cannot connect to it via chrome. 

In testing i've run ROS 1 on the same computer as the chrome extension, and it works with local host, but not with the IP address that my robot will be using - i.e 192.168.50.88

This functionality (ws://192.168.50.88:9090) works fine on the linux application, just not on the chrome extension, which as I say is required for my use case. 

Any ideas?

defunctzombie, 2022-08-11:
Unfortunately this is a limitation of the browser. When viewing an `https` website like `studio.foxglove.dev` the browser requires that other connections be "secure". Which means that you have to access the websocket server via `wss` instead of `ws`. However, to use `wss`, you'd need to setup a valid certificate for the ssl connection in front of your websocket server.

Chick92, 2022-08-11:
Hmmmm, is there a detailed guide on how to do that? 

Also, are there any plans for an Android version of the application?

defunctzombie, 2022-08-16:
There are no plans for an android version.

The guide for how to make a certificate would be specific to what setup you have. Unfortunately I don't have a one-size-fits-all guide. We use our cloud provider to manager certificates.

For now you'll need to use the desktop app to connect to rosbridge IPs for your robots.

closed 2022-08-16 by defunctzombie

Chick92, 2022-08-19:
Successfully implemented a certificate as you suggested - 

`sudo apt-get install ssl-cert
sudo make-ssl-cert generate-default-snakeoil
sudo usermod --append --groups ssl-cert yyuu
ls -l /etc/ssl/certs/ssl-cert-snakeoil.pem /etc/ssl/private/ssl-cert-snakeoil.key

sudo cp /etc/ssl/private/ssl-cert-snakeoil.key /etc/ssl/certs/
sudo chmod 644 /etc/ssl/certs/ssl-cert-snakeoil.key

modify rosbridge launch file and change following lines - 

<arg name="ssl" default="true" />
<arg name="certfile" default="/etc/ssl/certs/ssl-cert-snakeoil.pem" />
<arg name="keyfile" default="/etc/ssl/certs/ssl-cert-snakeoil.key" />`

This then allows foxglove studio chrome on my desktop (Ubuntu 20.04) to connect to my rosbridge (after i open another tab and go to https://192.168.50.88:9090)

Currently trying to replicate on an Android brower. 

Chrome, Edge and Samsung Browser (also chrome based) give the following error:

`(regl) "oes_texture_float" extension is not supported by the current WebGL context, try upgrading your system or a different browser`

The browser won't allow me to copy the full error message. 

This problem seems to bypassed by Chrome Beta, but foxglove studio won't connect - 

`Check that the rosbridge WebSocket server at wss://[192.168.50.88:9090](http://192.168.50.88:9090/) is reachable.`

This is then fixed by opening another tab as with before `https://192.168.50.88:9090`

Android tablets seem to have a preference for using the wifi rather than ethernet connection, so you need to open foxglove studio with wifi, then switch off wifi and put the connection details in. There's probably a way around this in the settings. 

the application is also not really that useable with a touch screen, but i'll try a mouse and update.

Chick92, 2022-08-19:
So using the chrome extension in android with a mouse is just like using it on a desktop. The lack of 3D rendering seems to just be bug that sorts itself out if you move the 2D render around a bit with the mouse and then click into the 3D.

defunctzombie, 2022-08-19:
> (regl) "oes_texture_float" extension is not supported by the current WebGL context, try upgrading your system or a different browser

Sounds to me like a graphics driver issue. The gl implementation supported by the browser on the device you are using doesn't support some specific feature the 3d panel uses. Give the `3d (beta)` panel a try and see if you experience the same problem.

Unfortunately the general fix for this is to get a device that does support the required WebGL and rendering api calls.

Chick92, 2022-08-19:
More than likely. It doesn't matter though, Chrome Beta seems to work fine and it's accessible via the play store. I'm curious as to why I have to open another tab and go to the websocket address though. My understanding of websockets is pretty limited. Not even sure that's the correct nomenclature. 

Anyway, it works so I'm happy. 

Cheers!