foxglove/studio #4123: Long time to establish data source connection

camchambers, 2022-08-12:
**Description**
When loading Foxglove Studio on our robots it takes an average of 1 minute 40 seconds to establish a data source connection and we think this relates to the rosbridge player topic initialization.

[image]

When reverting [this change](https://github.com/foxglove/studio/pull/3185/files#diff-502298948b3339b723b05e46807ad8b121945b7753732c899253f8aa37aceb6b) we noticed that the data source connection is established almost immediately. The PR notes that this change fixes an issue with the topic graph panel, but we don't find this feature useful. A typical environment for us consists of ~160 nodes, 980 topics, and 620 services.

Is there a workaround or another solution you could recommend?

- Version: 1.20.1 (also tested on 1.18.0)
- OS: Ubuntu
- Data source: rosbridge
- ROS distro (if applicable):

**Steps To Reproduce**
Load a production build of Foxglove Studio in an environment with a large number of nodes, topics, and services. 

**Expected Behavior**
The data source connection is established in a reasonable time.

**Actual Behavior**
The data source connection takes approximately 1 minute 40 seconds to establish. 

@jasonimercer

jtbandes, 2022-08-12:
Would you be willing to make a PR to fix the issue? It looks like the `Promise.all` behavior (making requests concurrently) could be restored, which might help in your case, while still keeping the bug fix intact.

jasonimercer, 2022-08-15:
The original issue seemed to have only expressed for a live ROS2 session so we'll need some help with review of the solution (we only have access to large systems running ROS1). We are exploring Promise.allSettled as a potential solution. This should allow as many of the nodes getting details as possible without triggering the early-out of Promise.all.

foxymiles, 2022-08-15:
@jasonimercer I can reproduce the original issue with a rosbridge session to my ros1 system running via Parallels here. I'm happy to help test if you have a fix in mind.

foxymiles, 2022-08-15:
@jasonimercer I'm not sure if it's easy for you to run studio from git but if so could you please give this proposed fix a try?

https://github.com/foxglove/studio/pull/4127

camchambers, 2022-08-15:
Thank you, @foxymiles . We'll test it out.

camchambers, 2022-08-15:
With this fix we now get a connection in about 30 seconds (down from the 100 seconds it was previously). An improvement, but not nearly as fast as reverting [this change](https://github.com/foxglove/studio/pull/3185/files#diff-502298948b3339b723b05e46807ad8b121945b7753732c899253f8aa37aceb6b) which connects immediately.

foxymiles, 2022-08-15:
Thanks for testing. The old behavior was broken for other use cases since it wasn't waiting for the promises to return so I'll need to think a bit more about other possible fixes.

camchambers, 2022-08-15:
No problem. Thanks for looking into this, and happy to do some more testing if needed.

foxymiles, 2022-08-15:
I've pushed some changes to that PR that I hope should speed things up for your case. Can you please try again when you have a chance?

camchambers, 2022-08-15:
Sounds good. I'll test the latest changes in the PR.

PR #4127 (2022-08-16): Run rosbridge topic graph setup queries in parallel
**User-Facing Changes**
Improves startup time of rosbridge sessions with large topic graphs.

**Description**
This runs the topic graph queries in parallel to speed them up.

Fixes #4123
closed 2022-08-16 by foxymiles