foxglove/studio #4130: Inconsistent font size in data source sidebar

jtbandes, 2022-08-15:
Font sizes are inconsistent for start/end time and duration in the "Data source" sidebar:

[image]

- Version: latest web
- OS: macOS Chrome
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): local mcap file

jtbandes, 2022-08-25:
Note that sometimes the font family also differs. Maybe some styles are not getting applied depending on the order they change in. Perhaps a precedence issue with our style libraries

foxymiles, 2022-08-26:
Fixed in https://github.com/foxglove/studio/pull/4273

jtbandes, 2022-09-01:
Still occurs on main:

[image]

foxymiles, 2022-09-01:
@jtbandes Do you have any repro steps? I'm not seeing it here.

jtbandes, 2022-09-01:
Seems to only happen when the start time is shown as a duration. Try this file:

[arrows.mcap.zip](https://github.com/foxglove/studio/files/9473795/arrows.mcap.zip)

By the way, it seems like a bug that we show the start time as 0.0 but the end time as a date in 1969.

osaro-imohe, 2022-09-02:
My pending PR resolves this issue.

foxymiles, 2022-09-06:
Closed by https://github.com/foxglove/studio/pull/4346

closed 2022-09-06 by foxymiles