foxglove/studio #4138: 3D beta in 2d mode cannot rotate and drag to change direction and position by mouse

lizhucheng, 2022-08-16:
**Description**
I have a layout with 3D beta panel, and this panel is 3D off. before I turn 3D on, I can not rotate the canvas and move the canvas to change the direction and position by mouse as what I can do in 3D mode. But after clicking then "3D" button twice, everything turns ok. 

- Version: 1.20.1
- OS: macOS 12.2.1, ubuntu focal 20.04. (It's probably nothing about system. )
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
- ROS distro (if applicable):

**Steps To Reproduce**

// a layout sample:
```
{
  "configById": {
    "3D!3udlufb": {
      "cameraState": {
        "perspective": false,
        "distance": 92.08032473661063,
        "phi": 0.7853981633974468,
        "thetaOffset": 0,
        "targetOffset": [
          0.07382587560197121,
          -0.24085274347973412,
          0
        ],
        "target": [
          0,
          0,
          0
        ],
        "targetOrientation": [
          0,
          0,
          0,
          1
        ],
        "fovy": 0.7853981633974483,
        "near": 0.01,
        "far": 5000
      },
      "scene": {},
      "transforms": {
        "base_link": {
          "visible": false
        },
        "map": {
          "visible": false
        },
        "sae_rear": {
          "visible": false
        },
        "sae_front": {
          "visible": false
        }
      },
      "topics": {},
      "layers": {},
      "publish": {
        "type": "point",
        "poseTopic": "/move_base_simple/goal",
        "pointTopic": "/clicked_point",
        "poseEstimateTopic": "/initialpose",
        "poseEstimateXDeviation": 0.5,
        "poseEstimateYDeviation": 0.5,
        "poseEstimateThetaDeviation": 0.26179939
      }
    }
  },
  "globalVariables": {},
  "userNodes": {},
  "linkedGlobalVariables": [],
  "playbackConfig": {
    "speed": 1
  },
  "layout": "3D!3udlufb"
}
```
**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-08-16:
Thanks for the report. This should fix it: https://github.com/foxglove/studio/pull/4155

lizhucheng, 2022-08-17:
> Thanks for the report. This should fix it: #4155

It's my pleasure. I'm very glad to hear that it has been fixed so quickly.

closed 2022-08-18 by jtbandes