foxglove/studio #4147: Remove un-actionable topic count from 3d (beta) sidebar.

defunctzombie, 2022-08-16:
**Description**

The 3d beta panel shows me some topic count in the settings and I have no idea what I am suppose to do with this information.

[image]

snosenzo, 2022-08-17:
Couldn't this be helpful for giving a user context as to whether they have topics that they want to see but might not realize are off?

defunctzombie, 2022-08-17:
> Couldn't this be helpful for giving a user context as to whether they have topics that they want to see but might not realize are off?

Those topics would show up in the topics section with a "closed eye" icon on them. From what I can tell the `0/7` in my data tells me that out of 7 total topics in my data, the panel is able to show me 0 of them. Not really that useful since I can already see what topics the panel supports because they would show up in the topics list.

PR #4182 (2022-08-18): Remove un-actionable topic count from 3d (beta) sidebar.
Fixes #4147

**User-Facing Changes**
 - Remove topic count from "Topics" node in 3D beta settings panel

**Description**
 - Remove topics count from 3D (beta) settings panel

#4147
commit 25dbc3e5 (2022-08-18): Remove un-actionable topic count from 3d (beta) sidebar. (#4182)

Fixes #4147
User-Facing Changes

Remove topic count from "Topics" node in 3D beta settings panel

Description

Remove topics count from 3D (beta) settings panel

#4147
closed 2022-08-18 by snosenzo