foxglove/studio #4156: Modifier+click to change step size in settings sidebar

jtbandes, 2022-08-16:
Each numeric stepper in the settings sidebar has a step value indicated by the panel. It would be nice if one could adjust the step size when clicking the arrows by holding down a modifier key.

For example, the Chrome devtools support this when changing values in the style inspector: https://developer.chrome.com/docs/devtools/shortcuts/

[image]

foxymiles, 2022-09-06:
This currently works for `shift` which gives you step * 10. Should we expand beyond that?

jtbandes, 2022-09-06:
I think at least having a way to do smaller steps would be nice.

foxymiles, 2022-09-06:
This would be very easy to add. We'd just need some consensus on what the key should be and how much it should reduce the step size.