foxglove/studio #4166: Is it possible to display additional positions on Map panel?

felixf4xu, 2022-08-17:
Hi,

Currently on Map panel, there is only one location which usually will be host vehicle/robot's position, can I add additional positions for example for remote vehicle/robot, so that there are multiple related locations displayed on the map at the same time?

(or any workaround in current studio version, image panel?)

defunctzombie, 2022-08-17:
The map panel will display all topics (each with their own color) that use the supported datatypes:

https://foxglove.dev/docs/studio/panels/map#supported-messages

closed 2022-08-17 by defunctzombie

felixf4xu, 2022-08-18:
Do you mean that I can send out multiple [foxglove.LocationFix](https://foxglove.dev/docs/studio/messages/location-fix) so that multiple location points can be displayed at the same time? I'm a little confused.

defunctzombie, 2022-08-18:
@felixf4xu yes - you can produce LocationFix messages on multiple topics.

felixf4xu, 2022-09-06:
multiple topics is not an ideal way in my case:

I need to display multiple locations of vary numbers, for example, 20 - 30 locations but it's dynamic.

In this case, I need to create 30 (the max possible number) topics, which is hard to remove/decrease dynamically to 20 locations.

```c++
      string serializeBuf;
      foxglove::LocationFix lf;
      lf.set_longitude(123.402413);
      lf.set_latitude(56.258867);
      lf.SerializeToString(&serializeBuf);
      server.sendMessage(chnFirstGps, nanosecondsSinceEpoch(), serializeBuf);   //many topics/channel IDs

      serializeBuf.clear();
      server.sendMessage(chnFirstGps, nanosecondsSinceEpoch(), serializeBuf);
```
sending out a empty string will not remove the location point from the map, that's why I said that it's not ideal.