foxglove/studio #4173: Performance degrades over time with certain datasets

foxymiles, 2022-08-17:
**Description**
Performance degrades over time recording certain topics.  Setting a follow window for the plot doesn't seem to solve the problem.

- Version: 1.21.0
- OS: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag
- ROS distro (if applicable): Noetic

**Steps To Reproduce**
Performance degrades over time recording certain topics. This was reported on a live session but we have a bag capturing the same topics.

See ilia-baranov-2022-08-17-10-08-24.bag and https://foxglove.slack.com/archives/C028UEY858S/p1660695147452789

iliabaranov, 2022-08-17:
Thanks for opening this! Happy to reply here with any additional detail, you guys rock!

iliabaranov, 2022-08-17:
https://user-images.githubusercontent.com/90713890/185204051-c1985caa-cf90-4ff2-a9aa-ac560f753732.mp4

Screen recording of the behavior, note that Foxglove is lagging behind both rqt and Formant interface

iliabaranov, 2022-08-17:
Interesting data point: seems to lose the lag momentarily when resizing any of the windows. Perhaps this is related to the fact that resizing wipes the graph

foxymiles, 2022-08-17:
@iliabaranov Yeah that could be. Thanks for the extra detail!

foxymiles, 2022-08-17:
@iliabaranov Can you please try one more thing? Remove the plot from your layout and run the same session with no plots and see if you see the same performance degradation?

iliabaranov, 2022-08-17:
Absolutely, trying now, will see the result in ~10 mins

iliabaranov, 2022-08-17:
Issue hasn't happened since! So seems to be plot related.

foxymiles, 2022-08-17:
Ok that's very helpful. Thanks for running that test.

defunctzombie, 2022-08-30:
Thanks for reporting this. After our call we fixed up the bad performance with having the data source sidebar open. The plot panel remains something that you need to be mindful to set a window size when viewing live data.

We have ongoing work to track performance in the app across devices and panels so that will further help track how performance changes in the app over time. I don't think there's anything else actionable from this bug report but if you do find yourself able to reproduce specific and egregious performance slow downs we are always happy to investigate.

closed 2022-08-30 by defunctzombie