foxglove/studio #4174: Request to add ability to change the title of any panel

iliabaranov, 2022-08-17:
Hello Foxglove team!

I'd love to be able to change the title of panels. In particular, I was thinking of the new gauges, as I want to have a few different ones, and it's hard to tell between them without labels. But the ability to change titles of other panels would also be very useful.

Thanks!

jessicaaustin, 2022-09-09:
+1 for this, Indicator has a similar problem. Would love to replace `Gauge` or `Indicator` with a title instead, like we can for plots

[image]

jtbandes, 2022-09-09:
@foxymiles @2metres can you provide some input on how we might implement this? Would the title be editable inline? Would this be part of the settings API or some separate API? Would the panel be able to change its own title or can only the user do it?

foxymiles, 2022-09-12:
> @foxymiles @2metres can you provide some input on how we might implement this?

I took a stab at an approach that bypasses the actual panel implementation entirely but it was not favorably received.

https://github.com/foxglove/studio/pull/3937

Is it necessary for the panel implementation to be aware of its title or can we implement in a general way that doesn't concern the panel internals?

jtbandes, 2022-09-12:
> Is it necessary for the panel implementation to be aware of its title or can we implement in a general way that doesn't concern the panel internals?

It seems more useful for a panel to be able to _set_ its own title than to read its own title. But neither of these seem very important and we could probably skip them for a first version of this feature.