foxglove/studio #4175: Incorrect looking transforms in 3D Beta

jasonimercer, 2022-08-17:
I have the 3D Beta view in visualizing in the map frame mode and I'm looking at a laser scan. The transforms from the map to this frame are:

```
(from /tf) { map -> odom -> base_link } -> (from /tf_static) { front_laser_link }
```

in the ROS1 bag that I'm playing, the `/static` transform with the front laser link transform doesn't arrive until 33s into the bag. Up to this point the old 3D view gives a transform missing warning and does not render the data until the tf arrives (see video between 8 and 9s)

https://user-images.githubusercontent.com/12039018/185213364-aaf1707a-bd64-454e-94d3-243ecbaf033f.mp4

Setting up a similar display with 3D Beta, a similar failure to transform is displayed as expected but once data comes it it seems like it is locking updates about an uninitialized origin (or something?) and the scans are not moving with the robot (see after 12s)

https://user-images.githubusercontent.com/12039018/185214139-5a59ea68-469e-4350-bcee-44a36047d2e6.mp4

I'll see if it's possible to sanitize the bag and reduce the size down to something that could be shared. It's from a testing facility so chances are good it can be scrubbed. 

- Version: 1.22
- OS:  `Linux 5.4.0-122-generic #138-Ubuntu`
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file
- ROS distro (if applicable): Ubuntu
- Chrome: 102.0.5005.115 

**Expected Behavior**
See old 3d video

**Actual Behavior**
See beta 3d video

jtbandes, 2022-08-19:
Thanks for the report. A sample bag we can use to reproduce the problem would be great.

defunctzombie, 2022-09-02:
So far I've not been able to reproduce this with the provided data or layout. We're happy to take a look and figure out any specific issues if you can send over the steps and data to reproduce.

closed 2022-09-02 by defunctzombie