foxglove/studio #4183: 3D visualisation does not appear to work on ARM64 architecture

Chick92, 2022-08-18:
Unable to visualise 3D point cloud data from a ROS1 source, using the Arm64 desktop app. 

I am able to view other topics in the message data field

Version: 1.22.0
OS: Raspberry pi OS
Data source ROS 1 native

defunctzombie, 2022-08-18:
@Chick92 Which 3d panel are you using? 3d (beta) or 3d (without the beta)?

jtbandes, 2022-08-18:
Can you please add some more details, such as what happens when you try to visualize the point cloud? Any errors visible in the app or the developer console? Can you share sample data (i.e. a bag) that we can use to reproduce the issue?

Chick92, 2022-08-18:
It was neither the 3d or 3d beta working, however after a reboot it seems to work fine... There were no errors visible. 

I'll try and do a more detailed bug report if it happens again. Unfortunately I'd already switched the device off before making the initial report. Sorry about that.

closed 2022-08-18 by Chick92