foxglove/studio #4219: 3d (beta) does not handle `renderDone` correctly

defunctzombie, 2022-08-19:
**Description**
The 3d (beta) panel calls render done in an effect on react render.
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ThreeDeeRender/ThreeDeeRender.tsx#L615-L617

However that is not when it has actually rendered the latest frame which happens on an animation frame call triggered here and run sometime later:
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ThreeDeeRender/ThreeDeeRender.tsx#L608-L610

The intent of calling `renderDone` is that you indicate to the message pipeline that you have rendered the frame of messages passed to you. This is how the pipeline keeps playback in-sync across the panels.

The 3d (beta) panel should be fixed to correctly call renderDone when the equivalent render frame has been rendered - in this case when the commands have been called for rendering the messages to the engine.

jhurliman, 2022-08-22:
Reclassifying as `ux` since this does not appear to degrade the app experience or break functionality in a user-facing way.

As far as fixing this, the current code looks like this:

```
  // Render a new frame if requested
  useEffect(() => {
    if (renderer && renderRef.current.needsRender) {
      renderer.animationFrame();
      renderRef.current.needsRender = false;
    }
  });

  // Invoke the done callback once the render is complete
  useEffect(() => {
    renderDone?.();
  }, [renderDone]);
```

`renderer.animationFrame()` synchronously calls `this.gl.render(this.scene, camera)` which is as close as we can get to "the frame has been rendered" with WebGL. Under the hood, it's just populating a frame buffer and the final compositing is asynchronously handled by the browser, but we don't have access to that level of feedback. I don't understand `useEffect` well enough to describe the situation reported in this ticket, where the second useEffect is called before or without the first useEffect. Can someone with a deeper useEffect understanding chime in here?

defunctzombie, 2022-08-23:
@jhurliman This needs fixing. I've put this as a bug and a blocker for new3d. The implementation does not follow what is expect of panels that handle new frame data outside their onRender functions.

defunctzombie, 2022-08-23:
I looked closer at some of the flow that happens after the onRender call from context. I agree its somewhat confusing with all the additional effects here. I do think we would greatly benefit from some cleanup of the logic here and am happy to pair on looking at options.

jtbandes, 2022-12-21:
Moved to Linear

closed 2022-12-21 by jtbandes