foxglove/studio #4221: Log panel should select topic automatically

rgov, 2022-08-21:
The log panel defaults to `/rosout_agg`. If there are no messages published on that topic in the data source, it should automatically select `/rosout`, otherwise it may appear that the log messages weren't recording which can be, from personal experience, a little alarming.

aschimmichanga, 2022-09-08:
This actually works as intended, since it's reasonable for Studio to choose one of the rosout topics if it exists, and the app doesn’t necessarily know in advance that the topic contains no data.

If you know in advance that the topic shouldn't be included, it might be better to exclude it from the data file as a workaround.

closed 2022-09-08 by aschimmichanga

rgov, 2022-09-08:
I'm not sure I follow you, but I do see that the code is supposed to default to the first available topic, or otherwise details to `/rosout`. Perhaps my layout had previously been initialized to use `/rosout_agg`?

jtbandes, 2022-09-08:
Yes, the app chooses a topic with the correct datatype. If your data source has two topics with that datatype, we don't have a way to know which one you really want. In your case if you don't have any data on a topic, it's probably better to just not record that topic.