foxglove/studio #4222: "Data went back in time" problem appears while scrubbing

rgov, 2022-08-21:
Foxglove Studio 1.22.0

If I scrub rapidly back and forth through a ROS1 bag file, a red (1) appears on the data source sidebar icon for a fraction of a second. The problem appears to be "data went back in time." I think this may be related to which parts of the file have been loaded in?

defunctzombie, 2022-08-21:
Adding a bug label because I don't think this should be happening and might be an indicator the message pipeline is not clearing as expected.

If you can reproduce this reliably could you record a video of it happening?

rgov, 2022-08-21:
https://user-images.githubusercontent.com/108767/185796292-b397f44a-8e29-4675-b12a-13817caef774.mov

amacneil, 2022-08-21:
It’s not wrong… the data did go back in time 😄

I assume Latching setting is enabled?

rgov, 2022-08-22:
I didn't know of this setting but it is enabled, although the issue seems to occur regardless of this setting.

PR #4266 (2022-08-25): Set _this.messages alongside _lastSeekEmitTime in stateSeekBackfill
**User-Facing Changes**
Quickly scrubbing back and forth does not cause a "Data went back in time" error to appear in data source problems.

**Description**
`stateSeekBackfill` would set `this._messages` after loading the backfill messages separately from setting `this._lastSeekEmitTime` which results in emitting a set of messages with no seekEmitTime making it look like the emit is a regular message playback rather than a seek. This would trip the message order tracker when a batch of older messages from a seek backfill (seek [... cut]
commit 2ced52bd (2022-08-25): Set _this.messages alongside _lastSeekEmitTime in stateSeekBackfill (#4266)

`stateSeekBackfill` would set `this._messages` after loading the backfill messages separately from setting `this._lastSeekEmitTime` which results in emitting a set of messages with no seekEmitTime making it look like the emit is a regular message playback rather than a seek. This would trip the message order tracker when a batch of older messages from a seek backfill (seeking to an older time) would be emitted without the seek emit time (because the user seeks again immediately and there's a next state which prevents  [... cut]
closed 2022-08-25 by defunctzombie