foxglove/studio #4224: Plot panel not resetting to correct range

defunctzombie, 2022-08-22:
**Description**

When I open the following data source (Foxglove HQ access required)

https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_Wm1gvryKJmREqnVT&ds.end=2022-08-22T00%3A02%3A14.251000000Z&ds.start=2022-08-21T23%3A41%3A31.514000000Z&layoutId=8ffa0ab6-3662-4248-a57c-3ba944f4b2d4&time=2022-08-21T23%3A51%3A11.458000000Z

I am able to plot this data using the following plot panel. However, the data shown is not the entire range.

[image]

I can manually zoom out to see the entire range, but when I click "reset view" I am again shown a smaller range.

[image]

foxymiles, 2022-08-22:
I think this may be fixed in https://github.com/foxglove/studio/pull/4145

defunctzombie, 2022-08-23:
I had a range of 10 seconds set on my plot panel so this was indeed showing me 10 seconds. 🤦

closed 2022-08-23 by defunctzombie