foxglove/studio #4228: transforms from tf_static not available anymore after time jump

d0my93, 2022-08-22:
Hi,

I am using foxglove desktop version 1.22 with ROS1 bag-files on ubuntu 20.04.
My bag contains 7 tf_static msgs. 
[image]

When I play back the bag without time jumps everything works fine and all static transformations are available.
However, when I jump in time only the last published static transformation is available and my transformation tree is not complete. 
So some Topics in the 3D panel are complaining about missing transforms. All tf_static msgs are published at the very beginning of the bag.

Main Question: How to use all the static transforms in foxglove even after time jumps?
How is tf_static handled by foxglove after time jumps? Is there a setting I am missing?

Thanks a lot in advance

jtbandes, 2022-08-22:
Hi, this is a known limitation. When seeking, we currently load only the 1 most recent message on each topic. To solve this problem we would like to implement "preloading" of all TF topics, so the 3D panel can access all TF data (including past and future data at any time step).

d0my93, 2022-08-23:
OK. Thanks for the quick respond

jhurliman, 2022-09-19:
Implemented in https://github.com/foxglove/studio/pull/4386

closed 2022-09-19 by jhurliman