foxglove/studio #4237: Upstream and release roslibjs changes

amacneil, 2022-08-23:
**Description**
We currently pin a specific commit in roslib:

https://github.com/foxglove/studio/blob/f67dc22be9ab51389d12333c94fe181f2dd9b5bb/packages/studio-base/package.json#L189

We should upstream our patches and release the changes.

jtbandes, 2022-08-23:
Upstreaming these changes would be tricky, we have made a bunch of changes to the build, see: https://github.com/RobotWebTools/roslibjs/compare/develop...foxglove:roslibjs:develop

defunctzombie, 2022-08-23:
I think we should take a different tactic - which is to grab just the websocket bits of roslibjs, port them to typescript, and make that a package:

https://github.com/foxglove/studio/pull/4240

jtbandes, 2022-08-23:
Should that package be in a separate repo (perhaps a new ROS JS monorepo which we can migrate our other packages to?) It doesn’t seem exclusive to Studio and would surely be useful for other people in the community who’ve struggled with the same import/bundling issues we did with roslibjs.

defunctzombie, 2022-08-23:
> Should that package be in a separate repo (perhaps a new ROS JS monorepo which we can migrate our other packages to?) It doesn’t seem exclusive to Studio and would surely be useful for other people in the community who’ve struggled with the same import/bundling issues we did with roslibjs.

Yea - I am using that PR to get a working version first and then the idea would be a separate repo.

jtbandes, 2022-09-08:
Something to keep in mind and possibly add in our new package: https://github.com/foxglove/studio/issues/4005

jtbandes, 2022-12-21:
Moved to Linear

closed 2022-12-21 by jtbandes