foxglove/studio #4241: Improve empty layout state

foxymiles, 2022-08-23:
**Description**
Our empty layout state could be improved.

- Version: studio.foxglove.dev
- OS: any

**Steps To Reproduce**
There are various ways to get into this empty layout state but it's a bit frustrating for the user because you have to go open the layout sidebar and select a layout or create a new one. Instead we could either add a "Create a New Layout" button here or just load the default layout.

[image]

snosenzo, 2022-08-26:
related to this issue?: https://github.com/foxglove/studio/issues/4100

defunctzombie, 2022-08-28:
I think it's reasonable to default to a layout with a single 3d panel, raw message panel, and logs panel. Between these three panels there is _some_ action you can take to see some data.

osaro-imohe, 2022-09-02:
I can't see that on studio desktop. Has this been resolved?

jtbandes, 2022-09-02:
No, it has not been resolved. One way to get to this empty state is to visit a url with `&layoutId=some-invalid-id`.

osaro-imohe, 2022-09-02:
I took a look at the code and have some thoughts. It seems there are three ways to see the empty layout state:

1. Not putting in a layout Id
2. Inputting an invalid layoutId
3. Not being able to retrieve the layout for a valid layoutId

The first situation was fixed by creating a new default layout when no layoutId was provided, then setting the new default  layoutId as the selectedLayoutId.

```
 const layout = currentLayoutId ? await layoutManager.getLayout(currentLayoutId) : undefined;
 if (layout) {
      await setSelectedLayoutId(currentLayoutId, { saveToProfile: false });
 } else {
      const newLayout = await layoutManager.saveNewLayout({
        name: "Default",
        data: defaultLayout,
        permission: "CREATOR_WRITE",
   });
   await setSelectedLayoutId(newLayout.id);
```

I considered applying the same solution to the other two instances that lead to the empty layout state. The issue is there's no way to know which layoutId is the default layoutId after you've already created the first one. For instance, if studio is loaded twice with no layoutId, it'd create two new default layoutIds. 

I think only allowing one default layout might be the way to go here. Once a default layoutId is created we could store that in localStorage and use that whenever a user runs into one of the three instances above.

So it'd be something like:
* load studio with no layoutId or invalid layoutId
* studio checks localstorage for default layoutId
* if no  default layoutId
     - studio creates a new default layout, then stores the default layoutId in localstorage
* else
     - studio sets the default layoutId as the selectedLayoutId

foxymiles, 2022-09-06:
Instead of trying to handle all these complicated cases maybe we should use add a "Create New Layout" button to the empty layout page?

osaro-imohe, 2022-09-06:
Okay sounds good, I'll handle that

foxymiles, 2022-09-06:
@osaro-imohe Before you try to implement that maybe we should see what @2metres and @amacneil think since they're sort of our UX gurus.

osaro-imohe, 2022-09-08:
Sounds good, I'll get that.