foxglove/studio #4242: [New3D] JointState support

jhurliman, 2022-08-23:
The end result should be the ability to load a URDF in the New3D panel and manipulate various parts of the robot using an incoming stream of JointState messages.

amacneil, 2022-09-23:
Related feature request: Foxglove-native JointState schema equivalent

amacneil, 2022-09-24:
Adding some more information:

In ROS, `JointState` messages are typically converted into `TFMessage` by [robot_state_publisher](http://wiki.ros.org/robot_state_publisher).

Rendering `JointState` without `TFMessage` (allowing `JointState` to override our regular TF graph) is useful in at least two situations:
1. The user is manually adjusting joint positions in Studio and not connected to a live ROS stack (i.e. paused bag playback, or URDF visualized with no other data source loaded)
2. The user has written joint state to a bag/mcap file and for whatever reason hasn't recorded TF messages (forgot to record TF, non-ROS robot, etc).

Per discussion with @jhurliman, overriding TF with joint states should probably be explicitly opt-in.